jeff-dh/SolidPython #1: preferred way to make blank part?

ipsod, 2021-06-24:
I've found it useful to start parts off with part().  This lets me comment/move/etc. the first item, just like I can the rest of them.  Since your fork doesn't support part(), what would be the preferred way to make a blank part?

jeff-dh, 2021-07-07:
I would suggest to use a "native OpenSCAD container" because that's what it actually is, right?

```python
>>> from solid import *
>>> a = union()
>>> a += cube(5)
>>> a += sphere(1)
>>> a
union() {
        cube(size = 5);
        sphere(r = 1);
}
```

If you don't want a `union`, you can do the same with a `difference` or `intersection`.

jeff-dh, 2021-07-07:
Btw: If you really really want or need to, you can import the legacy-extension and continue using `parts`. But I would strongly recommend to explicitly use `union,difference` and `intersection` instead like mentioned above.

```python
>>> from solid import *
>>> from solid.extensions.legacy import *
>>> p = part()
>>> p += cube(1)
>>> p += sphere(1)
>>> p
union() {
        cube(size = 1);
        sphere(r = 1);
}
```

(This might be a good solution to get (your) old code running without changes except for the import)

ipsod, 2021-07-07:
`union` makes more sense.  Thank you!

closed 2021-07-07 by ipsod