jeff-dh/SolidPython #2: Bring it back to master?

etjones, 2021-09-27:
Hi Jeff - 
   I've been away from SolidPython for a while, but I'm really impressed with the work you've done and it seems like you may have the energy to continue this project. Would you be interested in merging your stable features/refactoring back to master and taking on admin status there? I don't know that I'm looking to disappear entirely from the project, but I've been doing it for a decade and I've got a small kid these days, and it seems like you'd do a great job taking SolidPython forward. Any interest?

jeff-dh, 2021-09-27:
Hey Evan,

a couple of thoughts on this topic:

- I can imagine to maintain this branch (in terms of mainly bugfixes)
- I'm not sure whether I'm gonna invest a lot of time again to carry it further forward from this point on
- I don't think merging it with the master branch makes a lot of sense, because the code structure is too far apart
- I don't think it's possible to merge (single) features back
- I'm not really interested in spending time on packaging, but I might give it a try if necessary
- I can imagine to "link" it somehow back to SolidPython-"master"

So the question is what's the path to integrate it back.
The only solution that comes to my head would be kind of a new "major"-Release that switches to this branch in long term and could run in parallel in short term for kind of a beta phase. And I think it would make sense to start of from a blank "project page" because I think at least most of the issues and pull request are resolved in this branch. I don't know how that's possible on GitHub.

What do you think about these thoughts? Any other ideas?

jeff-dh, 2021-09-27:
> And I think it would make sense to start of from a blank "project page" because I think at least most of the issues and pull request are resolved in this branch. I don't know how that's possible on GitHub.

Ah, is it possible to link issues and pull requests to a certain branch?
If that's the case, we could simply pull this branch into the SolidCode repo and link issues and pull requests either to the master branch or to this branch and run both branches in parallel for a certain time.

etjones, 2021-09-28:
These are all valid 'next step' issues, but I think I could remove some or all of the book-keeping difficulties. I'll look at your code a little more, but I think you're right that this branch would replace the current `master` rather than try to merge into the old codebase. Packaging & release stuff is already set up in the original branch, and I don't *think* that would cause significant difficulties. Luckily, [Poetry](https://python-poetry.org/) makes packaging & releasing much much easier than it used to be in Python. 

So - I'm happy to look into some of the fine details; I'm most interested in incorporating your features & fixes, and getting an extra set of eyes on this code as far as maintaining go.

jeff-dh, 2021-09-29:
I had a look at poetry and I with you, that's a lot nicer than what I know from a decade ago....

I started forming a beta package, which seems pretty straight forward. The documentation needs some more attention ("merging" the README.rst). And we've got a problem with auto generated docs, because the builtins (which is 90% of the public interface) are generated dynamically and therefor don't even show up when sphinx tries to generate docs..... I don't have any idea what to do about it. Maybe just link to the openscad manual?

Even though I would like to not call it 2.0.0 (and instead 1.2.0), following semantic versioning I think we should call it 2.0.0. Furthermore I think the branch is in a state where it's fine to call it beta state, I think some people did already use it and there where no issues reported yet.....

From my side these are the few things to tackle before bringing it back to SolidPython-"master".

jeff-dh, 2021-09-30:
I started to work on the README.rst and packaging stuff:

https://github.com/jeff-dh/SolidPython/tree/master-2.0.0-beta-dev

I would continue to work on the master integration in that branch so I can leave the expSolid branch as kind of a stable branch for now and I have a dev branch for the intetragion.

It's work in progress!

etjones, 2021-09-30:
Excellent! As usual, I’m running to catch up.

jeff-dh, 2021-10-01:
So, I spent a little(!) time on the readme. I think it's now at least in a (rough) state where everything crucial is mentioned. I might change it further while we go, but I think for now it has to be good enough.

I would love to somehow get a couple of beta testers give it a shot and see what their feedback is. I think it would be great to do it before releasing it.
Do you see anyway to achieve it? I thought about simply creating an issue in the SolidPython issue tracker assuming that a couple of people are gonna get a notification and might have a look at it, what do you think about it?

etjones, 2021-10-01:
I think an issue tracking the branch makes good sense. I don't know anyone who's necessarily invested enough in the project to beta test for us, but it might be worth putting something up on /r/OpenSCAD to see if anyone's interested. 

Before we do that though, I wanted to run a couple things by you.  

The `.openscad` and `.implicit` files that you import to enable all the basic geometry functions is such a graceful way of bootstrapping into the target languages. I'm torn about that particular piece though, because it means that there's no explicit Python library that an IDE or static analyzer can use to know what the right arguments to, e.g. `cube()` are. In early versions of SP, I autogenerated the OpenSCAD classes, but somebody contributed explicit bindings with docstrings and everything, and that's been helpful to a lot of use using SP, especially once VS Code got better Python support. It's not nearly as graceful a solution as the runtime-analyzed files you've added, but would you be willing to let me incorporate the old explicit code?

I've also been pretty committed to maintaining unit tests for the package and keeping those valid. That's honestly been more work than I'm sure it's worth, but I'm a little nervous to just go without them. I'm not sure exactly where I'd start adding tests with the new format, but would you object to my adding some test infrastructure in a `test` dir?

All of the (great, clear, nicely numbered!) new example files now don't have a `#! /usr/bin/env python` line that would let them be self-executable (`01-basics.py` rather than `python3 01-basics.py`) and make some adjustments to `sys.path` that could clash with normal packaging behaviors. Are you committed to that approach? People have strong philosophical feelings for or against self-executing scripts, but I think not altering `sys.path` is a pretty common best practice these days.

jeff-dh, 2021-10-01:
Haha... I knew we where gonna get to this point ;)

Let me think about your points for a day or two. I'll come back to
it.

PS: vim is the one and only :-p

etjones, 2021-10-01:
No worries. I'm happy to talk through any of those issues. And it's totally fine at any point if you decide you want to back off and keep going in your own direction. I've been inspired by what you've done so far and I'd understand if you didn't want to dilute it with legacy concerns.

jeff-dh, 2021-10-01:
One is easy:

I don't mind putting a #!/bin/whatever at the top of the examples.

Altering the sys.path is "necessary" to be able to import the solid
package from within solid package (the examples directory) unless you
have it installed in some other sys.path directory.

Since I think it's important to be able to run the examples without
having to install solid in a regular sys.path directory I don't see any
way around it. I think it's important to be able to run the examples
from a github clone without installing it.

Furthermore, if you have a stable release installed in
.local/lib/python.../site-packages and are working on a development
version (or some other copy somewhere on your hard drive) running the
examples from within the dev version would import the solid package
from .local/lib/ptyon.../site-packages.

That's not acceptable for me and the reason why I alter the sys.path.
If you have any other solution for that issue let me know.

jeff-dh, 2021-10-01:
(you get more quick answers or better questions ;)

> The `.openscad` and `.implicit` files that you import to enable all
> the basic geometry functions is such a graceful way of bootstrapping
> into the target languages. I'm torn about that particular [...]

Do you have a solution how to support ImplicitCAD with the explicit
classes? Or would you drop the ImplicitCAD support?

I think it's impossible unless you do some harsh compromises.

etjones, 2021-10-01:
I hadn't thought about not installing the package. I think that would be an advantage for less technical users. 

It sounds like you may prefer not to work with virtual environments? I've found they're the best way to maintain my sanity with lots of different projects and dependencies around. In the virtualenv workflow, you have whatever version of `solid` installed in your current environment. Changing `sys.path` explicitly forces the example files to use _only_ the version of `solid` in the current directory tree. In effect, I think it's saying "I don't care how you have your environment configured, you should use this particular code", and that's why it's frowned on.

etjones, 2021-10-01:
I don't feel super strong about ImplicitCAD support. I messed around with it for a bit this spring because the auto-rounding features were nice, but I found it either slow or inaccurate in what it generated. Maybe it would have rewarded some more messing around. Among other things, ImplicitCAD doesn't support `polyhedron()`, so it was definitely a different way of thinking than some of what I'd been doing in OpenSCAD.

I think that getting IDE support for both OpenSCAD and ImplicitCAD, switched at runtime, would definitely be a nightmare. My priority-- and I understand if that's not your thing-- would be maintaining static analyzability for the core OpenSCAD language. If the ImplicitCAD bindings were dynamically generated like they are now, I think those definitions could be monkeypatched at runtime in a way that would maintain ImplicitCAD support. I'd be happy with that outcome.

jeff-dh, 2021-10-01:
I see.... as you might have already noticed I'm not really up to
date with the new shiny stuff (IPython, virtual envs, poetry...) and
catching up bit by bit.
All these topics where quite painfull or not even on the horizon
back than when I used to do programming extensively. C++98 and Python
2.6 where still up to date ;)

So I'm kinda out dated in some of these topics....

Aaeeh, yeah let me think about it (and maybe have a look at modern
virtual envs, what's the easy python way to use them for this issue?
Poetry?)

But what should be the point about using a different solid version for
an example than the one it shipped with? On the otherhand I don't like
the sys.path boilerplate code at the top either..... 

On Fri, 01 Oct 2021 14:37:19 -0700
Evan Jones ***@***.***> wrote:

> I hadn't thought about not installing the package. I think that would
> be an advantage for less technical users.
>
> It sounds like you may prefer not to work with virtual environments?
> I've found [...]

jeff-dh, 2021-10-01:
Concerning ImplicitCAD, I see.....

Let me think about it....

I don't think you're gonna get to a practical solution with ImplicitCAD
by monkey patching.... it's to different.

jeff-dh, 2021-10-02:
So, after sleeping over the issues I have a few more thoughts about it:

In general I'm ready to compromise on these topics, BUT..... ;)

I think we have to take a look at it in a little bigger picture:

First of all, I'll give you a picture of my design philosophy so that you might understnad my perspective. My design philosophy is a radical KISS approach combined with s strong emphasis on a clean architecture and this branch is designed according to it. The result is:

``` shell
  $ cat solid/core/*.py solid/extensions/*.py | grep -v '^$' | wc -l
  774
  $ cat solid/core/*.py | grep -v '^$' | wc -l
  557
```
Following my approach the whole core consists of ~600 sloc. I can -- and actually almost did -- manually test every single line of it while developing and thus I don't really need unit tests.
At the same time it enables you do get all kinds of features for free (e.g. ImplicitCAD support, the solid.utils features through bosl2).
The buttom line is: every single extra line of code needs to be maintained and increases the probability to introduce bugs.

Your objectives contradict with it. But I see the point that -- in contrast to me -- probably most of the people out there want to use VS or similar, code completion and all that stuff and therefor a mainstream library should somehow take care of it.

**hashbang & sys.path in examples**
I'm fine with your proposal

**unit testing**
well, as depicted it's not my style of developing, but as long as it's encapsulated -- and I don't have to maintain it -- I don't care

**explicit openscad builtins**
This point seems to me a little more complex, because it affects the design decision (the architecture) on a bigger scale.

As you can read in the current README, I don't like the idea of SolidPython providing a "modelling extensions library" by itself (`solid.utils` / the legacy-extension) and would like to get rid of it and replace it with `bosl2` as std library.

BUT, to be able to fully exploit bosl2 as std library / `solid.utils` replacement we would finally need to use their builtins! To be able to use the `bosl2-diff` or `bosl2-attachments` features you have to use their implementations of the primitives. Therefor I even think about replacing the regular OpenSCAD builtins with the bosl2 builtins / primitives by default. These would again be a dynamic set of builtins with the issues you mentioned.

So this comes for me to a decision whether SolidPython should be a slim core library (as in the expSolid branch) using bosl2 as std library to be a fully blown modelling library or whether SolidPython should continue to walk the path it does in the master branch up until now (providing and maintaining it's own "modelling extensions").

I can tell you clearly, I'm ready to take part in maintaining the `core` SolidPython library as it is in the expSolid branch, but I'm not gonna take part in maintaining `solid.utils` and all that stuff that comes along with it. Furthermore I don't like the idea of blowing up the core to accomplish needs that come along with it and am only ready to compromise up until a certain point on this topics (because otherwise we're gonna end up with a code base I don't like to maintain).

I would actually be ready to compromise on dropping the ImplicitCAD support and adding explicit OpenSCAD builtins (even though I would like to get rid of the type annotations), but I don't see how it fits the bigger picture in long term.....

Any ideas / thoughts about it?

jeff-dh, 2021-10-02:
One idea for a solution is on my mind:

We could split SolidPython into two parts:

SolidCore (which is basically the expSolid branch including explicit openscad builtins) and SolidUtils (which is everything else ontop of it to provide the full feature set of SolidPython).

The SolidPython package that would be released and distributed would be a meta package containing both modules. Or the otherway around, SolidPython could use SolidCore and provide the missing features (solid.utils).

In this design I would be ready to take care of SolidCore but not SolidUtils / SolidPython itself.

etjones, 2021-10-02:
Thanks for talking this through with me. I think that all sounds pretty doable. Here are my thoughts:

** 1) Core vs Utils** 
You know, I think you've convinced me about the minimal SP core.  I think using BOSL2 as a standard library is more in line with OpenSCAD and provides most if not all that the bloated current master does. I'm pretty set on maintaining explicit vs. dynamic code for OpenSCAD classes, but I think that's relatively straightforward and, while that would require a significant amount of code, there's next to no complexity there since it's only wrapper code & documentation. Getting rid of `solid.utils` reduces most of the code complexity and maintenance liability.

** 2) Separate meta-packages **
I think it's less confusing for users if there's only one SolidPython package. If there's any of the left-over code from `solid.utils` that I just can't bear to let go of, I might publish a separate package that's `solidpython_legacy` or  something, but I think your vision of a minimal central package is the way to go for the future, and it's reasonable not to want to maintain extra cruft. 

**3) ImplicitCAD **
If you're writing ImplicitCAD intentionally, you're probably using `radius` arguments or other non-OpenSCAD compatible code. If you're writing standard OpenSCAD and want to try it out in ImplicitCAD, can you just pass a `.scad` file to the ImplicitCAD program (or change the file extension?).  In either case, I think there's a relatively small use case that would call for using the *exact* same Python code, unchanged, to generate both OpenSCAD & ImplicitCAD code. I'd be happy to provide features for both languages but make the choice explicit in the Python code with either `from solid import *` or `from solid.implicit import *`.  Does that sound doable, or do you think the same-python-dual-target approach is something worth pushing harder for?

I'll work on a branch that embodies what we've been talking about and run it by you when that's ready. I'm optimistic that it wouldn't require huge changes to what's currently in ExpSolidPython.

jeff-dh, 2021-10-02:
>  I think using BOSL2 as a standard library is more in line with
> OpenSCAD [...]. I'm pretty set on maintaining explicit vs. dynamic
> code for OpenSCAD classes.

I would go along with that, but what about the BOSL2 primitives issue?

To really exploit it we would need to switch to the BOSL2 primitives,
e.g. in builtins.py replace any builtin primitives with:

```
	from solid.extensions.bosl2.primitives import *
	#or
	use("path/to/BOSL2/primitives.scad")
```

And we would be back to dynamic builtins.

Concerning ImplicitCAD: When I played around with implicitCAD it turned
out, that you can't switch between the two. A lot of primitives are not
available in ImplicitCAD and for some the parameters are different.

I don't see any other way to achieve it but to "bootstrap" from dynamic
builtins. But as mentioned earlier, I'm fine with just dropping it and
focusing on a pure OpenSCAD SolidPython.

jeff-dh, 2021-10-02:
Ah, hah!

The solution is already on the table:

We need explicit classes for the bosl2 primitives. I'll have a
look at it in the next couple of days, but I'm off for a day or so now.

Cheers.

jeff-dh, 2021-10-02:
A sketch for the next "version" of expSolid (based on BOSL2
support and incooperating your issues) is arising in my head. I hope
I'll find some time next week to have a look at it. It should not be
too much work.

etjones, 2021-10-02:
Just took a look at the BOSL primitives. Before you make any changes, check out the BOSL2 repository.  It looks like [this commit](https://github.com/revarbat/BOSL2/commit/935a113fcf20546618ffa3a4631c86beb7f03478) on 21 September replaced the `primitives.scad` file with calls to `shapes2d.scad` or `shapes3d.scad`.

jeff-dh, 2021-10-03:
Could you have a look at the current head of master-dev.... branch and check if code completion / static analyzers are working with it?

For now it's just a hack using the explicit classes from SolidPython master and conditionally(!) importing them. I don't know whether the static analyzers can handle it but if it works we would not have to drop the Implicit support.

jeff-dh, 2021-10-03:
IPython seems to handle it correctly. There's another problem in the convenience module at the moment for square and cube, I'm gonna have a look at it, but otherwise it seems to work, so I guess we can get BOSL2 primitives + implicit support.

jeff-dh, 2021-10-10:
Just to let you know, I didn't had any time to work on it yet and there will probably be no time for it in the next couple of weeks.

It's not canceled but might take some time.....

jeff-dh, 2021-10-10:
Btw: the issues you mentioned I think are all "fixed" in the current head. But the switch from OpenSCAD to BOSL2 primitives is not done and I think this should be done with care to get it right and from a - z to get a clean integration.

jeff-dh, 2022-06-07:
So, after spending some time recently again thinking and working on SolidPython / exp_solid, I decided that I don't want to replace the OpenSCAD primitives with the BOSL2 primitives. I'm afraid that could cause issues in some use cases and could cause quite some more work and / or would be a (too) drastic change.

There for I would propose that if we want to bring this branch back to master, this is probably the state we should take.

jeff-dh, 2022-06-10:
I spent a few hours on the bosl2 topic with the following results:

I updated and improved the bosl2 extension. It's now self contained in the solid package.

We should not replace the builtins with the bosl2 builtins, because for example `bosl2.cube` does not have the parameter `center` where as the default OpenSCAD builtin `cube` has that parameter. This would not be backwards compatible and I guess would cause trouble.

Buuut using SolidPython with bosl2 support is this easy:
```python
from solid.extensions.bosl2 import *
    diff() (
        cuboid(50) (
            sphere(d=40).attach(TOP).tag("remove"),
            cylinder(h=40, d=10).attach(CTR).tag("keep")
        )
    )
```

So I don't plan to make any major changes anymore.

Are you still interested to "Bring it back to master"?

Btw: I wondering whether it would make sense to release this as "solid2" package or similar. What do you think about it? Because it will break code. It's not 100% backwards compatible.

etjones, 2022-06-20:
Hi Jeff- Sorry for the delay. I am still interested in folding in your minimalist approach to SolidPython, although as you can see I've been distracted in other directions this last year.

I agree about separating the BOSL2 primitives from the OpenSCAD primitives; there's enough difference between them that I think it would be more confusing to users to have them combined rather than separate.

I'd be happy with a `solid2` package if you think that backwards compatibility would cause too much compromise with new features. At this point I think I've probably dragged my feet too long and I should just hand over the keys to you rather than slowing things down.

jeff-dh, 2023-03-08:
Soooooo, it seems that SolidPython2 is used be several people out there (https://pepy.tech/project/solidpython2), the feedback is pretty positive and it seems to work (a couple of minor issues where reported but nothing critical from my point of view, the pyright issue is the most "critical" one....).

So I would dare to call this fork "stable & production ready". Furthermore I'm fine with the "infrastructure setup" we have right now.

I was wondering whether you (@etjones ) would mind to place a "reference" to this fork on the SolidPython (master) page. Maybe a note in the Readme? The idea behind it is that people who search for SolidPython or OpenSCAD & Python probably end up at the SolidPython(1) page and if there would be a note in the Readme, they would be informed about this branch and can decide on their own which "version" they'd like to use.

From my point of view this would be the least expensive way to "bring it back to master".

Do you have any thoughts about it?

etjones, 2023-03-10:
I think that's great, Jeff. I'll put up a pointer at the top of the README on the SP page this weekend.

 And I've been dragging my feet on this for far too long. If we wanted to make your fork the Official SolidPython going forward, what would I have to do? I'm ready to just let it go where the energy is, with no code changes or nagging from me or anything else to get in your way. 

If I just say "Hey, this project is deprecated, look at SolidPython2 and install with `pip install solidpython2`", is that enough? I'd be willing to hand over the keys to the SP1 PyPi package too, but my only concern there is if there are any breaking changes so that if someone using SP1 updated to SP2 their existing code might break.

etjones, 2023-03-10:
I just put a note in at the top of the SP1 repo pointing to this repo and to the PyPI page. LMK if there's other information you'd like to get across there.

jeff-dh, 2023-03-10:
:thumbsup:

(I'm fine with the current "solution / setup" and close this ticket, reopen it if you have anything else)