jeff-dh/SolidPython #4: Convenience methods + operators are not recognized by pyright

elentok, 2022-08-30:
First, I just wanna say thanks for all the hard work, I've been using SolidPython for a while and loving it :-)

One thing that is a bit of an annoyance is many warnings when I use the pyright LSP, it doesn't recognize the monkey patched extensions to `ObjectBase`. See the screenshot:

[image]

I saw [this issue][1] on the pylance project (the VSCode extension that uses pyright) which seems to mean they're no going to support it probably :-(

I was wondering if have any thoughts/plans on this?

Thanks again,
David.

[1]: https://github.com/microsoft/pylance-release/issues/2335

jeff-dh, 2022-08-31:
I was already afraid something like that could pop up at some point.

I don't have any thoughts about it yet, but let me think about it for a couple of days, I'll try to come up with something....

jeff-dh, 2022-09-01:
checkout the branch mentioned above
(https://github.com/jeff-dh/SolidPython/tree/no_more_objectbase_patching)

I think it should fix your issue.

I'm not 100% convinced by it yet (because of some architectural issues) but I think that's the path to go.

I'll think about it a little more, but I'll probably gonna merge it into master soon.

jeff-dh, 2022-11-04:
Btw:

After thinking for it for a while now, I still don't like the archtectual design of the fix and therefor I'm not motivated to merge it into the master. At the same time I don't have any other idea how to fix this issue.......

I would leave this issue open for now and the branch with the fix as a separate branch.

commit a0928021 (2023-03-15): moved accessSyntax, operators and convenience support into core
	this should fix #4
	some refactoring in ObjectBase
	minor cleanup in ObjectBase
closed 2023-03-15 by jeff-dh