jeff-dh/SolidPython #7: How to view solid2 objects in Juypter lab

jreiberkyle, 2023-01-06:
I am able to view `solid` objects with `viewscad` in Jupyter lab but I can't figure out how to render `solid2` objects. Is there some change I need to make beyond replacing `solid.<geom>` with `solid2.<geom>`?

Note that running `viewscad.Renderer().render(solid2.cylinder(r=5, h=2))` has empty output while `viewscad.Renderer().render(solid.cylinder(r=5, h=2))` shows the 3d view. It appears that `solid2.cylinder(r=5, h=2).__repr__()` outputs scad code, which is useful but unfortunately not the 3d view.

[image]

Relevant package versions:
jupyter                  1.0.0
solidpython              1.1.3
solidpython2             2.0.0b1
viewscad                 0.2.0

jreiberkyle, 2023-01-06:
Ok with some investigation it appears the limitation is with viewscad. Makes sense, the repo hasn't been updated in 4 years. I was able to get viewscad and solidpython2 working by just updating viewscad's dependency to solidpython2 and then importing solid2 as solid in my (hacked) working version [here](https://github.com/jreiberkyle/ViewSCAD). If my reasoning is correct, you may want to consider updating the docs to indicate that solidpython2 and the viewscad pypi package are not compatible.

jeff-dh, 2023-01-07:
Yeah, I agree about updating the readme.

Do you have a proposal? I guess mentioning your branch and.... (I've never been in touch with viewscad at all).

If you like to make any suggestions and create a PR.......

jeff-dh, 2023-01-08:
I added two sentences to the readme summarizing your results and linked your fork and this issue.

My support for this topic is limited, since I don't use jupyter notebooks at all. Suggestions, PRs,... are welcome!

closed 2023-01-08 by jeff-dh

kylekyle, 2023-10-14:
I love designing STLs using Jupyter/SolidPython in VSCode, so I'd be interested in making this work. 

I think viewscad should just be baked into solid2. ViewSCAD is a single python file. If you just replace all instances of `solid` with `solid2`, it works. Then we can simply add a `#render` method `OpenSCADObject`.  If @jeff-dh is interested in a pull request, I can make one.

jreiberkyle, 2023-10-14:
@kylekyle you should check out [jupyterscad](https://github.com/jreiberkyle/jupyterscad) - just `pip install jupyterscad`. Works like ViewSCAD but supports SolidPython2, integrating external stl's, and a bit more control over visualization. Check out the bottom of the README for how to move from ViewSCAD to jupyterscad. I need to talk to @jeff-dh about switching the recc over from my viewscad fork to jupyterscad

jeff-dh, 2023-10-14:
@jreiberkyle 
https://github.com/jeff-dh/SolidPython/wiki/Features#jupyter-renderer

like this ok or would you like anything else / more?

jeff-dh, 2023-10-15:
We could add a minimalistic example to the SolidPython2 docs / wiki:

Install JupyterSCAD:
```
pip install jupyterscad
```

```python
from solid2 import cube
from jupyterscad import render

obj = cube(1, 2, 3)
render(obj)
```

That's all, right?

PS: can't test the code because I don't use jupyter notebooks at all....

Btw: the jupyterscad README uses `import jupyter_scad` instead of `import jupyterscad` (I guess that's outdated, right?)

kylekyle, 2023-10-15:
Jupyter SCAD is exactly what I was looking for! Don't know how I missed it. Thanks for pointing me in the right direction! 

An example in the wiki is a good call since at the moment a Google search for "solidpython jupyter" only returns information about viewscad.

jreiberkyle, 2023-10-15:
@jeff-dh that looks great. BTW what is the "Lite" for in JupyterLab Lite Notebook Support?

@kylekyle I'm glad you are finding it useful! I just recently created the project so I can understand why it isn't yet very discoverable. I expect the primary audience will discover it through SolidPython2 docs.

jeff-dh, 2023-10-15:
On Sun, 15 Oct 2023 14:49:13 -0700
Jennifer Reiber Kyle ***@***.***> wrote:

> @jeff-dh that looks great. BTW what is the "Lite" for in JupyterLab
> Lite Notebook Support?

While having a look at it I came across JupyterLite (and tried to make
solidpython and jupyterscad work in it, but unfortunately but obviously
openscad is missing....)

https://jupyter.org/try-jupyter/lab/

"JupyterLite is a JupyterLab distribution that runs entirely in the
browser built from the ground-up using JupyterLab components and
extensions."

I'm gonna remove it.

jreiberkyle, 2023-10-15:
> Btw: the jupyterscad README uses import jupyter_scad instead of import jupyterscad (I guess that's outdated, right?)

ahh! great catch! fixed here: https://github.com/jreiberkyle/jupyterscad/pull/7

jreiberkyle, 2023-10-15:
JupyterLite looks very cool! It makes me wish for something like OpenSCAD As A Service. An alternative web notebook interface that allows for installing openscad is google colab. Throws an interesting error I'll have to look into but still renders and provides interactivity.

[image]