jeff-dh/SolidPython #8: set_global_fn can only be set once.

mcfaddendavid, 2023-01-27:
First off, thanks for the great work. I recently got round to switching to solid2 and have been very impressed.

I use Jupyter notebooks to design parts interactively. To keep the Openscad's 3D preview responsive, I use the default (or lower) `$fn `value. That's usually sufficient for designing, and once it's done I crank up the `$fn` and call openscad to render the STL. When tweaks are required I might have to set it back. Rinse and repeat. All of this is within one Python session.

The recommended way for doing this is to use `set_global_fn(128)`

This works the first time. But the second time this function is called, I get

`ValueError: Multiple instances of ScadVariable with the same name.`

Thankfully this works again since the` file_header=` argument in `scad_render_to_file `was restored. But I really think that `$fn` parameter ought to be able to be overwritten, and not just set once. Maybe this can be changed?

Thanks again for the great work.

jeff-dh, 2023-01-29:
I guess I fixed it. Give it a try.....

closed 2023-01-29 by jeff-dh

mcfaddendavid, 2023-01-30:
Works great!
Thanks for all the good work.

commit fae56593 (2023-01-29): builtin vars (fn, fa, fs, vpd, ...) can be set multiple times (fix #8)