jeff-dh/SolidPython #11: Support operations on BOSL2 directional vectors

jreiberkyle, 2023-03-13:
BOSL2 includes [directional vector constants](https://github.com/revarbat/BOSL2/wiki/constants.scad) such as CENTER, BOTTOM, LEFT, etc. These vectors can be added together to specify a corner, e.g. `anchor=BOTTOM + LEFT`.

solid2 imports these constants as `OpenSCADConstant` instances and these instances cannot be combined, so `BOTTOM + LEFT` would fail. It would be nice to be able to combine them to restore the corner functionality.

There are two ways I can think of:
1. change `extensions/bosl2.py` to include these vectors as globals along with their values, e.g.
```python
# BOSL2 anchors as arrays so we can add them together
TOP = np.array([0,0,1])
BOTTOM = np.array([0,0,-1])
LEFT = np.array([-1,0,0])
RIGHT = np.array([1,0,0])
FRONT = np.array([0,-1,0])
BACK = np.array([0,1,0])
CENTER = np.array([0,0,0])
```
2. Implement operations in the `core/object_base/OpenSCADConstant` class e.g.:
```python
    def __str__(self):
        return self.name

    def __add__(self, other):
        if isinstance(other, OpenSCADConstant):
            return OpenSCADConstant(self.name + "+" + other.name)
        else:
            return self + OpenSCADConstant(str(other))

    def __sub__(self, other):
        if isinstance(other, OpenSCADConstant):
            return OpenSCADConstant(self.name + "-" + other.name)
        else:
            return self - OpenSCADConstant(str(other))

    def __truediv__(self, other):
        if isinstance(other, OpenSCADConstant):
            return OpenSCADConstant(self.name + "/" + other.name)
        else:
            return self / OpenSCADConstant(str(other))

    def __mult__(self, other):
        if isinstance(other, OpenSCADConstant):
            return OpenSCADConstant(self.name + "*" + other.name)
        else:
            return self * OpenSCADConstant(str(other))
```

jeff-dh, 2023-03-13:
I like your second approach more even though I have a question about it. Why do you check whether other `isinstance(...OpenSCADConstant)`? Did you encounter any case where that check was necessary? Is there any case where you need to apply an operator to an OpenSCADObject and a OpenSCADConstant?

Take a look at `solid2/extensions/greedy_scad_interface/scadVariable.py:ScadValue` I guess that's pretty much what your second approach is about, right? I already have a local copy (and I can push it to a branch) where I merged ScadValue and OpenSCADConstant and it seems to be working, but I'm not sure whether that would cause issues somewhere else....

Otherwise, I'm fine with your first approach, even though -- as soon as I get rid of my concerns ;) -- I would prefer the second approach because it's more generic.

jeff-dh, 2023-03-13:
I'm actually against the `isinstance` check, because what about this:

```
cube(SOME_CONSTANT * 3)
```

This would work without the check -- at least with the `ScadValue` version of your second approach -- but would not work with the check, right? So the question to me really is, why did you put it in? :-p

This is a great issue, thx!

jeff-dh, 2023-03-13:
Check out the mentioned branch and test it whether it behaves the way you'd expect it.

jeff-dh, 2023-03-13:
aaah, I see why you added the isinstance check..... take a look at the ScadValue solution, it handles that case implicitly.

jreiberkyle, 2023-03-13:
I really like that `OpenSCADConstant` is now available in the `solid2` scope, so I can create any constant I want, aka `OpenSCADConstant('2*TOP')`.

The change to `OpenSCADConstant` itself takes care of all of my use cases without the need for making a custom `OpenSCADConstant`, which is great! I tried: `TOP + LEFT` and `2*INCH` and they worked.

jeff-dh, 2023-03-14:
You're ought to use `scad_inline(...)` for that case actually,
which was imported by default before.

But I like the idea of adding the operators to OpenSCADConstant so we
can do it the python way.

jeff-dh, 2023-03-15:
So, I merge the mentioned branch above into master.

Note: I reverted the `from object_base import *` to a selective import and the result is that `OpenSCADConstant` is not exposed at module / package level again. But you can either use `scad_inline` (which is kind of the public interface to it) or import OpenSCADConstant yourself (`from solid2.core.object_base import OpenSCADConstant`).

closed 2023-03-15 by jeff-dh