jeff-dh/SolidPython #12: Any ideas for how to make this library's operators play nicely with linters?

ipsod, 2023-03-15:
Right now, this code: `cube([10, 20, 30])  + sphere(10)` gets this error from pyright:
`Operator "+" not supported for types "cube" and "sphere"  Operator "+" not supported for types "cube" and "sphere" Pyright (reportGeneralTypeIssues)`

I looked, but couldn't find an obvious way to fix it.

ipsod, 2023-03-15:
I've fixed it by modifying ObjectBase in object_base.py to inline the dunder methods that were added in operators.py, and add typing.  There's repetition here, where there wasn't there, but copy/paste worked, so I'd rather break DRY-principles than break my editor.

``` python
class ObjectBase:
    def __init__(self):
        self.children = []

    def __add__(self, x) -> ObjectBase:
        """
        This makes u = a+b identical to:
        u = union()(a, b )
        """
        res = builtins.union()

        #add self or all its children to res
        if isinstance(self, builtins.union):
            for c in self.children:
                res.add(c)
        else:
            res.add(self)

        #add x or all its children to res
        if isinstance(x, builtins.union):
            for c in x.children:
                res.add(c)
        else:
            res.add(x)

        return res
    def __or__(self, x) -> ObjectBase:
        """
        This makes u = a+b identical to:
        u = union()(a, b )
        """
        res = builtins.union()

        #add self or all its children to res
        if isinstance(self, builtins.union):
            for c in self.children:
                res.add(c)
        else:
            res.add(self)

        #add x or all its children to res
        if isinstance(x, builtins.union):
            for c in x.children:
                res.add(c)
        else:
            res.add(x)

        return res
    def __radd__(self, x) -> ObjectBase:
        """
        This makes u = a+b identical to:
        u = union()(a, b )
        """
        res = builtins.union()

        #add self or all its children to res
        if isinstance(self, builtins.union):
            for c in self.children:
                res.add(c)
        else:
            res.add(self)

        #add x or all its children to res
        if isinstance(x, builtins.union):
            for c in x.children:
                res.add(c)
        else:
            res.add(x)

        return res

    def __sub__(self, x) -> ObjectBase:
        """
        This makes u = a - b identical to:
        u = difference()(a, b )
        """
        res = builtins.difference()

        if isinstance(self, builtins.difference) and len(self.children):
            for c in self.children:
                res.add(c)
        else:
            res.add(self)

        res.add(x)
        return res

    def __mul__(self, x) -> ObjectBase:
        """
        This makes u = a * b identical to:
        u = intersection()(a, b )
        """
        res = builtins.intersection()

        if isinstance(self, builtins.intersection) and len(self.children):
            for c in self.children:
                res.add(c)
        else:
            res.add(self)

        if isinstance(x, builtins.intersection):
            for c in x.children:
                res.add(c)
        else:
            res.add(x)

        return res
    def __and__(self, x) -> ObjectBase:
        """
        This makes u = a * b identical to:
        u = intersection()(a, b )
        """
        res = builtins.intersection()

        if isinstance(self, builtins.intersection) and len(self.children):
            for c in self.children:
                res.add(c)
        else:
            res.add(self)

        if isinstance(x, builtins.intersection):
            for c in x.children:
                res.add(c)
        else:
            res.add(x)

        return res

    def __invert__(self):
        return builtins.debug()(self)

    def add(self, c):
        def _add(c):
            assert(hasattr(c, "_render"))
            self.children += [c]

        if isinstance(c, list):
            for cc in c:
                _add(cc)
        else:
            _add(c)

        return self

    def copy(self):
        return deepcopy(self)

    def _render(self):
        s = ''
        for c in self.children:
            s += c._render()
        return s

    def __call__(self, *args):
        #translate(...)(cube())
        #this adds cube() to translate.children
        for a in args:
            self.add(a)
        return self

    def __repr__(self):
        return self.as_scad()

    def as_scad(self):
        from .scad_render import scad_render
        return scad_render(self)[:-1]

    def save_as_scad(self, filename='', outdir=''):
        from .scad_render import scad_render_to_file
        return scad_render_to_file(self, filename, outdir)

    def save_as_stl(self, filename='', outdir=''):
        from .scad_render import render_to_stl_file
        return render_to_stl_file(self, filename)

```

jeff-dh, 2023-03-15:
This is a duplicate of #4, right?

Since we got two tickets about this issue now, I think I finally need to come up with something.

I would be fine with your solution, but your solution only covers a fraction of the problem. It's not only the operators but also everything from the `convinience` extension. Further more it questions the overall extension design (as soon as you extend `ObjectBase` from an extension pyright & co don't work properly anymore...).

jeff-dh, 2023-03-15:
I merged the proposal I did back than for #4 into master -- even though I'm not 100% happy with it, but well.....

So I guess this issue should be fixed now. The PyPi package is not updated yet, I would like to have a couple of you give it a try first.

If you're still having issues, please reopen this issue....

ipsod, 2023-03-15:
I'm getting an error with this code:

``` python
a = cylinder(r=10, h=120, center=True).rotate(0, 90, 0)
```

`Cannot access member "rotate" for type "cylinder"  Member "rotate" is unknown Pyright (reportGeneralTypeIssues)`

ipsod, 2023-03-15:
I don't see a way to re-open.  Might be something you can enable on your end?

ipsod, 2023-03-15:
A similar error with this code:

```python
window_cube = cube(200, 55 ,50, center=True).down(10)
```

`Cannot access member "down" for type "cube"  Member "down" is unknown Pyright (reportGeneralTypeIssues)`

jeff-dh, 2023-03-15:
I reverted the merge for now and gonna have a look at it soon.....

jeff-dh, 2023-03-15:
So I just tested it a little more and it seems to be working "pretty"
good. The lines you mentioned don't cause errors over here!?!

But I had some issues at the beginning too. I think(!) pyright needs
some time to realize and rescan the new sources. At least I think it was
an issue for me.

If you can, try to somehow clear the pyright cache or somehow trigger
that pyright rescans the whole library. (i.e. I had to make a dummy
change in builtins.py for some reason so it picked up cube....!?!)

ipsod, 2023-03-16:
I checked out `git+https://github.com/jeff-dh/SolidPython.git@no_more_objectbase_patching`, and you're right, it's fixed!

Sorry, I should have assumed it was my own fault for a little longer.  I must've updated the wrong venv or something.

ipsod, 2023-03-16:
I had some issues getting it running, but I installed with `pip install -e git+https://github.com/jeff-dh/SolidPython.git@no_more_objectbase_patching#egg=solidpython2`, and I think restarting my venv was the final touch, and it seems to be working correctly now.  I'm no longer getting any of the errors described above.

I ran through the examples to check for errors.   I did get lots of errors on solidpython2/solid2/examples/07-libs-bosl2-attachable.py

Not sure if further examples have errors.

jeff-dh, 2023-03-16:
Ok, I reverted the revert ;)

Concerning BOSL2: Yeah, I know about it and I don't see any reasonable approach to fix it. The point is the following:

When I refactored SolidPython I did it with a pretty "dynamic approach". For example the OpenSCAD builtins itself where loaded dynamically from a *.scad file using the same mechanism as if you would import any regular scad file (this is still the case for the implicitCad builtins, take a look at `core/builtins.py` if you're interested). This works pretty good and makes the code pretty slim. Buuuuuut..... the result was that all the code analyzers (lsps, linters,...) didn't work.
Therefor we replaced the dynamic builtins with the explicit builtins from SolidPython1. Now we had more or less the same issue with the operators and the access style syntax. In both cases extra code was moved into the core package of SolidPython2 and it gets more and more crowded in there (which I don't like ;). But anyway, there seems to be no way around.

For BOSL2 -- which gets loaded dynamically like any other scad import -- we have the same issue. If you take a look at the `extensions/bosl2.py` extensions you'll see, that the whole bosl2 extension is ~50 lines of code and the actual bosl2 support code -- without convenience functions -- is ~30 lines of code. This comes at the cost that static analyzers can't figure out what's going on there -- because everything gets loaded and the module itself gets populated at runtime.
The only alternative I see would be to explicitly define every single(!) bosl2 module! This is no options for me for several reasons:

- it would be a whole lot of work
- we would need to adjust to every interface change in bosl2
- where are we gonna stop to add explicit support? What about `mcad`, `bosl(1)`,......, `your-personal-favorite-lib`?

So the bottom line for me is at the moment, that we should hopefully have a SolidPython2 package where all the "core-features" play nicely with static analyzers now, but we have to make a trade off and stop at a certain point otherwise we will end up with a huge amount of "dead code" just to satisfy static analyzers and I'm not willing take that unless someone comes up with a design -- or something else -- that somehow keeps it out of the core module!

To give you numbers what we're talking about:

The whole `core` module has ~1736 lines of code. 829 lines of them are only there to satisfy static analyzers!!! We could immediately get rid of ~700 lines of code (the whole `core/builtin_primitives.py` file) if we would drop support for static analyzers!

If anyone comes up with a solution how to satisfy static analyzers from a SolidPython2-extension, I don't care at all what's actually going on in the extensions itself -- especially since we can out source extensions.

jeff-dh, 2023-03-16:
Well, after another thought about it: At least major parts of the bosl2 "issue" could be fixed in the bosl2-extension, but we would still probably add another few thousands lines of code. I'm not gonna do it, if anyone wants to do it, feel free to create a new -- static -- bosl2 extension!

ipsod, 2023-03-16:
Might bosl2 static analyzer support be a job for stubs?  I think that stubs are what people use when a library doesn't generally support static typing, but I don't know anything else about stubs or anything about the complexities of bosl2.

I've had really good luck using TabNine to write SolidPython code, this week.  I guess that maybe geometry is so generic and common that it just really knows what I'm doing before I do it, with practically every line of code.  I was writing a generic wrapper class for objects, and I think I wrote fewer than 1/3 of the actual lines of code (since many things had to be repeated several times, for forward/back/left/etc., code generators can knock that stuff out).  If writing the interface would be more rote data-entry than actual thinking.

Another option might be to write a stub generator (?).  It could just find function definitions in bosl, and convert those to a Python equivalent?  It seems like one of those things where it would be really easy to get an 80% solution, but maybe not a 100% solution.  So, it might need an additional tool, something like a set of tests, maybe with a simple GUI that lets the user accept or reject the result (since some tests would be hard to do programmatically).  You could use an sqlite database to store the results of each test, and only require re-validation on tests where the signature has changed - PonyORM is old, but it makes setting up an sqlite database for something like that practically zero-cost.

So, yeah, as someone who doesn't know much, I'd vote for some simple tooling to help automate creation and maintenance of the bosl2 extension (and any other openscad extensions).  I'm planning to build some tooling soon that might help in doing something like the running the user through a series of GUI pass/fail tests.  I don't think it'd be difficult to parse an scad file and look for function definitions.

jeff-dh, 2023-03-16:
Well, that's almost exactly what SolidPython2 does...... the only thing missing is the code generation. And that's a good idea I didn't had in my mind before.

SolidPython includes a ScadParser that extracts exactly the information you need to generate the stubs (I think the comments / doc string are missing). From this information SolidPython generates SubClasses of `OpenSCADObject` and that's how `extensions.bosl2.cube` gets formed. You could also take the output of the parser and generate python stubs from it.

Feel free to implement it ;) let me know if you really head towards it, I could provide you with several hints where to pick up parts you would need.

About the complexity: the OpenSCAD-Builtins stubs are 750 lines of code and we're talking about maybe 30 builtins. I guess BOSL2 is a loooot bigger (BOSL2 has > 50 scad files, each containing several modules (maybe 20 each?), so we end up with maybe 1000 stubs......). I guess your idea about generating the stubs is feasible, but I'm not sure whether I'll have motivation to spend time on it. Maybe.... but.... maybe not... ;)

ipsod, 2023-03-16:
Ok, sounds good.  It's in that same box for me, maybe, maybe not.  I may be more inclined than you, since it matters to me, and the work I'm planning might make it a good learning step.

jeff-dh, 2023-03-16:
slightly modified, but more or less copy & paste from `solid2/core/parse_scad.py:74-83`:
```python
        from ..libs.py_scadparser import scad_parser
        modules, functions, global_vars = scad_parser.parseFile(filename)

        callables = []
        for c in modules + functions:
            args = [p.name for p in c.parameters if not p.optional]
            kwargs = [p.name for p in c.parameters if p.optional]

            callables.append((c.name, args, kwargs))

        generateStubs(callables)
```

Implement the `generateStubs` function and extend the parser to extract the doc strings that would pretty much be it. Right now I'm a little pissed -- by accident by PonyORM (is there anyway to express "select ... AS my_alias from ..." in PonyORM generator syntax?) -- and might(!) have a look at it....

ipsod, 2023-03-16:
A stub generator would be useful beyond bosl2, wouldn't it?  There are lots of libraries out there.  That's one of the huge advantages of OpenSCAD, over other code-cad programs.

I don't know, about PonyORM.

jeff-dh, 2023-03-17:
Just to keep you up to date: I got a proposal for a bosl2 stub generator in my local copy. Seems to work. Needs a little more cleanup / renaming. I might commit something soon.

ipsod, 2023-03-17:
Cool.

I had trouble that bosl2 "infected" the global namespace, even though I imported it with `import bosl2 as b2` and kept this gear generator from functioning.  Is there a way to keep the namespaces separate?  Maybe I'm being lazy by asking you, and should research OpenSCAD more, but I wouldn't know where to start and don't have much time right now.

jeff-dh, 2023-03-17:
Are you talking about Python namespaces or openscad namespaces?

ipsod, 2023-03-17:
openscad namespaces.

I have this code:
``` python
import solid2.extensions.bosl2 as b2

PG = gears.PolyGears()
h = 10
draft = False
wigwag = PG.wig_wag(h=h, lh=1 if draft else 0.2, cycles=1)
part = PG.spur_gear(h=h, helix_angle=wigwag, n=64, center=CC)  # , chamfer_shift=45, chamfer=1)
d = 2.5 * inch
```

and openscad console outputs these warnings and makes the output wrong (maybe just makes it the default output of that function with no args?):
```
Compiling design (CSG Tree generation)...
[WARNING: variable add not specified as parameter in file PlayToday.scad, line 33](https://github.com/jeff-dh/SolidPython/issues/33,/gk/shapes/output/PlayToday.scad)
[WARNING: variable chamfer not specified as parameter in file PlayToday.scad, line 33](https://github.com/jeff-dh/SolidPython/issues/33,/gk/shapes/output/PlayToday.scad)
[WARNING: variable chamfer_shift not specified as parameter in file PlayToday.scad, line 33](https://github.com/jeff-dh/SolidPython/issues/33,/gk/shapes/output/PlayToday.scad)
[WARNING: variable ded not specified as parameter in file PlayToday.scad, line 33](https://github.com/jeff-dh/SolidPython/issues/33,/gk/shapes/output/PlayToday.scad)
[WARNING: variable helix_angle not specified as parameter in file PlayToday.scad, line 33](https://github.com/jeff-dh/SolidPython/issues/33,/gk/shapes/output/PlayToday.scad)
[WARNING: variable m not specified as parameter in file PlayToday.scad, line 33](https://github.com/jeff-dh/SolidPython/issues/33,/gk/shapes/output/PlayToday.scad)
[WARNING: variable n not specified as parameter in file PlayToday.scad, line 33](https://github.com/jeff-dh/SolidPython/issues/33,/gk/shapes/output/PlayToday.scad)
[WARNING: variable type not specified as parameter in file PlayToday.scad, line 33](https://github.com/jeff-dh/SolidPython/issues/33,/gk/shapes/output/PlayToday.scad)
[WARNING: variable x not specified as parameter in file PlayToday.scad, line 33](https://github.com/jeff-dh/SolidPython/issues/33,/gk/shapes/output/PlayToday.scad)
[WARNING: variable z not specified as parameter in file PlayToday.scad, line 33](https://github.com/jeff-dh/SolidPython/issues/33,/gk/shapes/output/PlayToday.scad)
```

But, commenting out the line `import bosl2...` gets rids of the warnings.

jeff-dh, 2023-03-17:
so I just pushed a bosl2 extensions that works pretty good with pyright over here. Could you give it a try and verify that it's working?

https://github.com/jeff-dh/SolidPython/tree/bosl2_stubs

ipsod, 2023-03-17:
I'm still getting an error on `examples/07-libs-bosl2-attachable.py`.  Fewer errors, but still...

``` python
cc = cubic_barbell(100)(
        cube(50, center=True).attach(TOP),
        sphere(50).attach(LEFT),
        show_anchors(30)
    )
```
```
 │     Cannot access member "attach" for type "cube"  Member "attach" is unknown Pyright (reportGeneralTypeIssues) [17, 31]
 │     Cannot access member "attach" for type "sphere"  Member "attach" is unknown Pyright (reportGeneralTypeIssues) [18, 20]
```

jeff-dh, 2023-03-17:
yep, I won't / "can't" fix it! Everything else (except the two access
style functions) should work.

jeff-dh, 2023-03-17:
That seems like an openscad issue.

Try to reorder the imports in your python file, that might fix it.

```python
# from
from solid2.extensions.bosl2 import *
[...]
polyGear = import_scad("path/to/polygear")

# to
polyGear = import_scad("path/to/polygear")
[...]
from solid2.extensions.bosl2 import *

# or the other way around
```

This should change the order of the includes / uses in the scad file and that might fix your issue.

ipsod, 2023-03-17:
I think I noticed a side effect,t hat I had to update this code.  Not sure if it's relevant to you, or not - it's fine with me.

```python
    it = b2.skin(flats, z=[0, l], slices=10)
    # now must be:
    it = b2.skin.skin(flats, z=[0, l], slices=10)
```

ipsod, 2023-03-17:
Changing the import order did fix it.  Thanks again!

jeff-dh, 2023-03-17:
The `b2.skin.skin` issue is fixed in the `bosl2_stubs` branch....... that was indeed a bug

ipsod, 2023-03-17:
Ok, the fix works for me.

ipsod, 2023-03-17:
Oh, wait, no... changing the import order does not seem fix the issue .  The import order is corrected in the .scad file, but it doesn't seem to change the effect.

Even if I modify the code to do the calculations "before" the bosl imports, it still doesn't work:

```
use </gk/shapes/PolyGear/PolyGear.scad>;
use </gk/shapes/scad/arcs.scad>;

translate(v = [0, 0, -5.0000000000]) {
	translate(v = [0, 0, 5.0000000000]) {
		spur_gear(add = 0, backlash = 0.1000000000, chamfer = 0.0000000000, chamfer_shift = 0.0000000000, ded = 0, helix_angle = [0.0000000000, 1.8837155859, 3.7599970069, 5.6214394376, 7.4606966149, 9.2705098312, 11.0437365805, 12.7733787470, 14.4526102231, 16.0748038494, 17.6335575688, 19.1227196925, 20.5364131779, 21.8690588226, 23.1153972833, 24.2705098312, 25.3298377651, 26.2892004013, 27.1448115740, 27.8932945766, 28.5316954889, 29.0574948339, 29.4686175219, 29.7634410394, 29.9408018528, 30.0000000000, 29.9408018528, 29.7634410394, 29.4686175219, 29.0574948339, 28.5316954889, 27.8932945766, 27.1448115740, 26.2892004013, 25.3298377651, 24.2705098312, 23.1153972833, 21.8690588226, 20.5364131779, 19.1227196925, 17.6335575688, 16.0748038494, 14.4526102231, 12.7733787470, 11.0437365805, 9.2705098312, 7.4606966149, 5.6214394376, 3.7599970069, 1.8837155859], m = 1.0000000000, n = 64, pressure_angle = 20.0000000000, type = 1, x = 0, z = 10.0000000000);
	}
}

include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/version.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/constants.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/transforms.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/distributors.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/mutators.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/color.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/attachments.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/shapes3d.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/shapes2d.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/drawing.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/masks3d.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/masks2d.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/math.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/paths.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/lists.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/comparisons.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/linalg.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/trigonometry.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/vectors.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/affine.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/coords.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/geometry.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/regions.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/strings.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/skin.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/vnf.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/utility.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/partitions.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/wiring.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/linear_bearings.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/bottlecaps.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/fnliterals.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/beziers.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/screws.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/polyhedra.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/gears.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/rounding.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/turtle3d.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/structs.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/hingesnaps.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/threading.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/tripod_mounts.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/sliders.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/screw_drive.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/nema_steppers.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/walls.scad>;
[... cut ...] k/venv/src/solidpython2/solid2/libs/BOSL2/joiners.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/cubetruss.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/modular_hose.scad>;
include </home/zero/rockbox/work/venv/src/solidpython2/solid2/libs/BOSL2/metric_screws.scad>;
```

I don't really understand OpenSCAD, or its "declarative" nature, or however they describe it.  It doesn't flow linearly.  The includes and such are no exception.

jeff-dh, 2023-03-18:
I think I got something for you:

- update to the latest bosl2_stubs branch head
- install solid2_extensions (`pip install solidpython2-extensions`)
- use PolyGear with pyright support (`from solid2_extensions.PolyGear import PolyGear, PolyGearBasics, PolyGearUtils`)

I have a `stub_generator` now that can generate a `solid2` extension from any(?) OpenSCAD library. So let me know which libraries you're using and I can add them to the solid2_extensions easily.

ipsod, 2023-03-18:
Auto-completion looks great.  The output is weird, though.   Adding `from solid2_extensions.PolyGear import PolyGear, PolyGearBasics, PolyGearUtils` produces most of this output:

[image]

ipsod, 2023-03-18:
Also, congrats on putting that together so quick.  That was fun to be a part of.  Good work.

jeff-dh, 2023-03-18:
The "weird stuff" should be fixed if you update your
`solidpython2-extensions` package.

jeff-dh, 2023-03-18:
Sooooooo, I merge everything into master, updated the pypi packages and I guess............ we're finally done with this ticket -- and made some good steps forwards, yeah! PyRight for everything :D --, right?

I'll close this ticket, if there are any issues, let me know.

closed 2023-03-18 by jeff-dh

ipsod, 2023-03-18:
Ok, looks good. Thanks!

Is there any way to limit what gets imported with `import solid2.extensions.bosl2`? 

It seems that bosl2's gear library's functions clobber PolyGear's, when I import both.

jeff-dh, 2023-03-18:
unfortunately no.

You could comment out `solid2.extensions.bosl2.__init__.py:6` and then
manually import what you need. Or take a look at
`solid2.extensions.bosl2.all.py` and adjust according to your needs.

This is a little unlucky, because the bosl2 extensions was designed
some time ago and I don't want to make any breaking changes....
otherwise we could....... ;)

jeff-dh, 2023-03-18:
Now, there is a way ;)

update `solidpython2-extensions` to the brand new `0.1.3` version it
includes a extension called `BOSL2` which does not import anything by
default.

I just made it the default behaviour for all generated libraries, that
they do not import anything by default. This also changed in the
PolyGear library.

I think this makes sense, so you can pick very selective what you
actually want.

The `bosl2` extensions stays unchanged, because it's the `stdlib` of
solid2 and I'm not gonna make any breaking changes to it.

ipsod, 2023-03-18:
This new thing has the `bosl2.skin` >> `bosl2.skin.skin` thing.  However, it works!

You're ridiculously fast.  I wish I could know what you're doing to know if I'm just slow or if you have some sort of elite workflow that could be replicated.

ipsod, 2023-03-18:
Just a note, if you're using any libraries in the new BOSL2, you have to also `from solid2_extensions.BOSL2 import std`.

It might be nice if importing any BOSL2 library also imported std.

ipsod, 2023-03-18:
On the other hand, maybe it's good to keep things explicit.  Neither of the above are complaints - just mentioning, in case they're of concern to you.

Thanks for the library and the upgrades!

jeff-dh, 2023-03-18:
yeah I know....

[blablablabla] ;)

that was now too fast over here..... the `BOSL2` extensions is unmature.

I need to think about it a little.... there's another issue right now:
`solidpython2_extensions.BOSL2.std` is empty! So you can't use `cuboid`
in solidpython even though you imported `BOSL2.std`. You would need to
explicitly import `BOSL2.shapes3d` (or similar) to use it....

Hmmmm I need to think about all of that for some time and gonna propose
a soution soon.

jeff-dh, 2023-03-19:
So, the result is the following:

I further improved `solid2.extensions.bosl2` and I did make a breaking change in it!

`solid2.extensions.bosl2` now only imports `bosl2.std` by default. Everything else needs to be imported manually:

```python
from solid2.extensions.bosl2 import *
from solid2.extensions.bosl2 import metric_screws
```

I think this should fix your issue, right? It's not in the pypi package yet but in the master-2.0.0-dev branch.

I'm probably gonna remove the `solidpython2_extensions.BOSL2` extension again.

jeff-dh, 2023-03-20:
Anything further should go into #16

Btw: solidpython2_extensions.BOSL2 is fixed and I'm probably gonna leave it in solidpython2_extensions to have a bare BOSL2 lib available if necessary.