jeff-dh/SolidPython #16: Improved BOSL2 -> BREAKING CHANGES!

jeff-dh, 2023-03-20:
Hey everybody,

I dared to make some (hopefully) minor breaking changes to the bosl2 extension. The reason is that a new idea came up from @ipsod which completely changed the game of how to integrate 3rd party libraries.

The result is that we now have full pyright and access syntax support for bosl2!

The down side are the breaking changes. The breaking changes are the following:

**imports**
by default `solid2.extensions.bosl2` does not import the whole library anymore. It only imports `bosl2.std` by default. So if you're missing any sub modules you have to manually import them:
```python
from solid2.extensions.bosl2 import *
from solid2.extensions.bosl2 import gears, beziers, screws, cubetruss, metric_screws
```

**pure bosl2**
The bosl2 library is now designed in such a way, that it is completely independent from the regular builtins. Thus you don't have access to the regular builtins by default any more.
The most important ones are porbably `translate` and `rotate`. You should now use `move` and `rot` from bosl2 instead.
If you need some of the regular builtins you have to manually import them (`from solid2 import translate`), but be aware, that this will break bosl2 access syntax support for the resulting objects

There we're a few more minor (maybe breaking) changes but I hope no one will notice them, because they where -- always -- meant to be internal.

I'm sorry for any inconvenience, but I think it's worth the price to get the `bosl2` extensions straight and a couple of major steps forward.

I hope you agree with this decision.

If you have any issues, let me know!

PS: These changes are -- up until now -- only in the master-dev.... branch and not in the pypi package. I would really appreciate it if some of you (@ipsod, @jreiberkyle, @elentok, ?) would give it a try and give feedback before I package it into the pypi package. 
Thanks in advance!

PPS: Thanks a lot to @ipsod for coming up with the idea of generating stubs. This is the basis for all the improvements that were added the last couple of days.

ipsod, 2023-03-22:
Everything looks good here.  I've switched over, everything has worked, and I haven't had any unexpected behavior.

Congrats, and thank you!

Where should we put requests for extensions/stubs?  I'd like an Archimedean spiral module, like this: https://cults3d.com/en/3d-model/various/archimedean-spiral-module

jeff-dh, 2023-03-23:
> Where should we put requests for extensions/stubs?

-> #15

jeff-dh, 2023-03-23:
> I'd like an Archimedean spiral module, like this: https://cults3d.com/en/3d-model/various/archimedean-spiral-module

You can also try to create (a local) one for yourself, especially for such small -- probably not wide spread -- libraries.

Go to "any" directory and run the following:
```bash
cd your/lib/dir
../path/to/solid2/extensions/openscad_extension_generator.py ../../relative(!)/path/to/openScad/lib libName
```

This should create a new directory containing the python stubs that you should be able to import exactly the same way as the PolyGear libs.

Since using that lib requires to include the original .scad file in the resulting scad file, the `../relative/path/to/openscad/lib` must not change relative to the python package! The `solidpython2-extensions` lib simply always keeps a copy of the openscad lib next to the python module.

closed 2023-04-21 by jeff-dh