jeff-dh/SolidPython #18: Support BOSL2 screws.screw_hole

ssuchter, 2023-04-11:
This code doesn't work:

```
screws.screw_hole("6-32", length=height / 25.4)
```

It gets this error message:

AttributeError: module 'solid2.extensions.bosl2.screws' has no attribute 'screw_hole'. Did you mean: 'screw_head'?

ssuchter, 2023-04-11:
It seems like the problem is that the BOSL2 code is pasted into this repo, with an old version.

Is this the desired practice? Or should this be using another mechanism, like git Submodules?

jeff-dh, 2023-04-11:
Should be fixed (updated to latest BOSL2 HEAD + integrated as submodule)

ssuchter, 2023-04-11:
Works now, thanks.

I see that you re-added things as files, instead of as a submodule. If I could make it work with a submodule (but still installing a snapshot of BOSL), would you like a PR for that?

jeff-dh, 2023-04-11:
Yeah, sure.

ssuchter, 2023-04-12:
Question about your workflow: When you build the pip package do you do it via 'poetry build' or via 'python -m build' or something else?

jeff-dh, 2023-04-12:
I use `poetry build`.

ssuchter, 2023-04-14:
Here you go: https://github.com/jeff-dh/SolidPython/pull/19

ssuchter, 2023-04-19:
Closing this issue, as BOSL2 has been updated in master-2.0.0-beta-dev

closed 2023-04-19 by ssuchter