jeff-dh/SolidPython #24: Fix doubled arguments generated in bosl2_generator.py

ssuchter, 2023-04-20:
The raw output of bosl2_generator.py generates some incorrect output. For example, at the moment, this code is produced in bosl2/distributors.py:

```
def __init__(self, path=None, n=None, spacing=None, sp=None, dist=None, rotate_children=None, dist=None, closed=None, p=None, **kwargs):
       super().__init__("path_copies", {"path" : path, "n" : n, "spacing" : spacing, "sp" : sp, "dist" : dist, "rotate_children" : rotate_children, "dist" : dist, "closed" : closed, "p" : p, **kwargs})
```

The argument dist=None is listed twice in the __init__ method declaration and in the call to super().__init__. This is a bug.

jeff-dh, 2023-04-20:
The issue is, that this seems to be "legal" in OpenSCAD but not in Python. The OpenSCAD code really contains these double parameters.
For now I just deleted the second occurrence of the parameter. This works fine, but it actually changes the signature of the corresponding function. Up until now I couldn't come up with a better solution.

If that's the solution, we could take care of it in https://github.com/jeff-dh/SolidPython/blob/9482e85596cef17e580f6c4b72a89154c0bc0743/solid2/extensions/openscad_extension_generator.py#L41

 I did not had a closer look at it: why are there double parameters in openscad at all? I assume for convenience, but I did not had a closer look at it, what's the semantic behind it?

jeff-dh, 2023-04-20:
Btw: this is not `bosl2_generator.py` but `openscad_extensions_generator.py`

ssuchter, 2023-04-20:
I think it's non-sensical to have double parameters in OpenSCAD, I just don't think there's any validation. I tested this code:

```
module draw_cube(a, b, a) {
    cube([a, b, a]);
}

draw_cube(1, 2, 3);
```

It produced a rectangular prism with sides 1, 2, and 1. So while the later duplicate arguments can be specified, they cannot actually be read inside the function. (As I'd kind of expect...)

Since we want solid2 to be robust to whatever upstream BOSL2 feeds it, I think the right thing to do is to change the code generation to match the manual fixes you've had to do.

| Btw: this is not bosl2_generator.py but openscad_extensions_generator.py

Understood, I was just focusing the issue report on the entry point rather than the root cause. Thanks, though.

jeff-dh, 2023-04-21:
But what about this:

```openscad
module box(a=0, b, c=0, b)
{
    if(a != 0)
        cube(a, b);
    else
        cube(c, b);
}

box(a=1, 2);

translate([0, 0, 10]){box(c=3, 4);}
```

jeff-dh, 2023-04-21:
But what about this: 

https://github.com/revarbat/BOSL2/blob/bfa5fbc81aa2cbdc177e839f68baa62d1ae09b57/distributors.scad#L1480

jeff-dh, 2023-04-21:
I added a line to the genrator(s) to remove duplicates from the parameter list.

so the genrator(s) do the formerly manual tweaking now.

So I think we can close this issue, right?

closed 2023-04-21 by jeff-dh