jeff-dh/SolidPython #27: prebuild.py call causes "build error"

jeff-dh, 2023-04-21:
https://github.com/jeff-dh/SolidPython/blob/4337922c5535f9d564bd7ac15d733e98f6f3a948/pyproject.toml#L45

For some reason this causes this error over here when I call `poetry build`:

```$poetry build
Preparing build environment with build-system requirements poetry-core>=1.0.0
Building solidpython2 (2.0.1)
Traceback (most recent call last):
  File "/[..]/SolidPython/setup.py", line 2, in <module>
    from setuptools import setup
ModuleNotFoundError: No module named 'setuptools'

Command '['/tmp/tmpkz3ceeao/.venv/bin/python', '/[..]/SolidPython/setup.py', 'build', '-b', '/[..]/SolidPython/build']' returned non-zero exit status 1.

```
Note: setuptools is installed

changing the pyproject.toml file like this:

```python
[tool.poetry.build]
generate-setup-file = false
script = "prebuild.py"
```
makes the call work again, but the build process creates a different file!?! `solidpython2-2.0.1-cp310-cp310-manylinux_2_35_x86_64.whl` instead of `solidpython2-2.0.1-py3-none-any.whl`

@ssuchter Any idea what's going on there?

jeff-dh, 2023-04-21:
maybe related:

https://github.com/python-poetry/poetry/pull/5401

ssuchter, 2023-04-21:
> python-poetry/poetry#5401

Hm, I don't think we need the setup.py file, so how about we just keep the
generate-setup-file = false line?

I saw that in the examples I looked at (having the generate-setup-file =
false) but it worked in my setup, so I didn't think it was important and
opted for a shorter change. But if it's breaking your setup, I vote for
just fixing it by including the option, without figuring out why.

jeff-dh, 2023-04-21:
Well but as mentioned, the output is different. Why is the wheel
file a different (at least the name which seems to be less portable,
at least the name says so).

ssuchter, 2023-04-21:
Hm, I need to try to reproduce your problem. I'll try.

jeff-dh, 2023-04-22:
Do you have all necessary information?

If you remove the whole `[tool.poetry.build]` section, it creates a *py3-none-any.whl file (as it used to be). With the prebuild.py call and `generate-setup-file = false` it produces *cp310-cp310-manylinux_2_35_x86_64.whl which is according to [this](https://peps.python.org/pep-0491/#file-name-convention) less compatible.

Does it behave the same on your system?

ssuchter, 2023-04-22:
I'll take a look tonight (California time)

ssuchter, 2023-04-23:
Ok, I've reproduced the issue, looked through poetry code, and found that for some bizarre reason, poetry doesn't support using a build script and producing a pure python wheel. Others complain about the same issue:

https://github.com/python-poetry/poetry/issues/2051

I'm trying a workaround from that issue now.

ssuchter, 2023-04-23:
I think the lack in underlying poetry is hard to work around. I think the right thing to do is to not use the prebuild script. Originally, it was being used to autogenerate stubs, but you changed my mind on that. Now it's just validating that the submodules got initialized, but we can use the tests to validate that now. I'll make a PR for this trivial change.

ssuchter, 2023-04-23:
Here's the PR for you: https://github.com/jeff-dh/SolidPython/pull/28

closed 2023-04-23 by jeff-dh