jeff-dh/SolidPython #30: 07-libs-bosl2-attachable.scad uses absolute path in includes

ssuchter, 2023-04-23:
This file has the last calling user's home directory in its path:

test/examples_scad/07-libs-bosl2-attachable.scad

include </home/jeff/code/solid_stuff/SolidPython/solid2/extensions/bosl2/BOSL2/version.scad>;
include </home/jeff/code/solid_stuff/SolidPython/solid2/extensions/bosl2/BOSL2/constants.scad>;
...

I don't think this should be checked into git as-is, as it creates spurious test failures for any user who isn't jeff, with their git clone in code/solid_stuff/SolidPython

jeff-dh, 2023-04-23:
hmmmm this is too bad!

If we would exclude that file (and all the others which contain includes) from the tests, we would exclude all bosl2 tests...... which was why we started this whole testing stuff ;) Even though the testing is a good think independently from bosl2, I would still like to have at least one bosl2 file be considered in the test.

Don't know how we can achieve it yet, gonna think about it for some time....

jeff-dh, 2023-04-23:
This could be a solution:

Only make the diff call / check if "bosl2" is not part of the filename. This would disable the diff check for bosl2 examples but would still call the openscad render check.

I think this is not an optimal solution, but the best I could come up with up until now.

jeff-dh, 2023-04-23:
Or something similar to this:

```python
git diff -U0 | grep '^[+-]' | grep -Ev '^(--- a/|\+\+\+ b/)' | grep -v "^.include" | grep -v "^.use
```
https://stackoverflow.com/a/26622262

This would exclude all includes / uses from the diff check, but since afterwards openscad render is called, that should check whether the includes are good.

ssuchter, 2023-04-23:
Different idea: how about rewriting the include line to be relative during
the file copy step?

jeff-dh, 2023-04-23:
Checkout the latest head. I did rewrite the includes to relative includes and it seems to work. Would you give it a try on your system?

ssuchter, 2023-04-24:
I just tried it on my system, and it didn't do what I wanted. I think it's an artifact of it picking up the installed instance of solid2, rather than the development version:

```
diff --git a/tests/examples_scad/07-libs-bosl2-attachable.scad b/tests/examples_scad/07-libs-bosl2-attachable.scad
index 3712fe1..d4a548b 100644
--- a/tests/examples_scad/07-libs-bosl2-attachable.scad
+++ b/tests/examples_scad/07-libs-bosl2-attachable.scad
@@ -1,31 +1,31 @@
-include <../../solid2/extensions/bosl2/BOSL2/version.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/constants.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/transforms.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/distributors.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/mutators.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/color.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/attachments.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/shapes3d.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/shapes2d.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/drawing.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/masks3d.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/masks2d.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/math.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/paths.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/lists.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/comparisons.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/linalg.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/trigonometry.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/vectors.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/affine.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/coords.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/geometry.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/regions.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/strings.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/skin.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/vnf.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/utility.scad>;
-include <../../solid2/extensions/bosl2/BOSL2/partitions.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/version.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/constants.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/transforms.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/distributors.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/mutators.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/color.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/attachments.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/shapes3d.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/shapes2d.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/drawing.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/masks3d.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/masks2d.scad>;
+include <../../../../../../opt/homebrew/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/math.scad>;
```

jeff-dh, 2023-04-24:
Did you run the test in a virtual env (-> poetry shell)?

ssuchter, 2023-04-24:
Ah, that fixed it. I might made some helpful docs (and maybe an error message in run_tests.py) so that the next person doesn't make the same mistake I did.

ssuchter, 2023-04-24:
BTW, after getting it working inside the venv, I experimented with commands outside the venv. This works for me:

```
poetry build
poetry install
poetry run python tests/run_tests.py
```

closed 2023-04-24 by ssuchter

jeff-dh, 2023-04-24:
I'm not really convinced about adding checks for it because:

- any work within a development (solidpython) package works that way (`poetry shell` / `poetry run`). If you want to run the examples from a development package you also need to run them like that. Same for using it inside a python shell. Therefor -- at least for me -- it's obvious to work that way and I guess it's the "poetry way" of doing it -> poetry docs.
- how would you like to check it? I guess you would need to compare the filename of `__main__` with the file name of the imported `solid2` package, right? And this would polute every script (tests, extensions generating scripts, examples, ...).
- these are the basics of virtual environments...... I would assume they should simply work (as they do with poetry) and everyone should get familiar with it

What I could image is to add some (development) documentation. How do you work with a development version of solidpython? How do you run the tests? How do you build the package? ....

If you like, go for it. I would suggest either a section in the readme or an extra md file somewhere (root or docs).

ssuchter, 2023-04-24:
Definitely development documentation comes first. Yes, I'll draft
something. Probably in a separate md file.

Sean