jeff-dh/SolidPython #31: Trouble using bosl2's `skin` module

ipsod, 2023-05-02:
I'm having trouble with bosl2's `skin` module.
 
``` python
# minimal example
import solid2.extensions.bosl2 as bosl2
from solid2 import *

hole_template = circle(d=10)
part1 = hole_template
part2 = scale(0.9)(hole_template)
def draw(parts):
    it = bosl2.skin(parts, z=[0, 1], slices=2)
    save(union()(it))

# does work:
draw([part1, part1])

# does not work:
# draw([part1, part2])
# openscad error:
#   ERROR: Parser error: syntax error in file PlayToday.scad, line 40
```

It seems like `union()`, etc., may be disallowed in the skin module?  Maybe they cannot be allowed due to OpenSCAD argument syntax?

Maybe the solution is to create an OpenSCAD module, to get around syntax limitations?  Is there a way to create an OpenSCAD module from SolidPython, so that I can do that?

jeff-dh, 2023-05-02:
```python
>>> from solid2.extensions.bosl2 import *
>>> hole_template = circle(d=10)
>>> part1 = hole_template
>>> part2 = scale(0.9, p=hole_template)
>>> skin([part1, part2], z=[0, 1], slices=2).save_as_scad()
```

The issue seems to be, that you can't pass a module as a parameter to another module. The issue is in the definition of `part2` and if you take a look at the 6th example of the [skin docs](https://github.com/revarbat/BOSL2/wiki/skin.scad#functionmodule-skin) they use transformation functions instead. This seems to work in solidpython as well as depicted above.

jeff-dh, 2023-05-03:
Btw: There is a way to create openscad modules from Solidpython2:

```python
>>> from solid2.extensions.bosl2 import *
>>> from solid2_extensions.exportResultAsModule import exportResultAsModule
>>> 
>>> @exportResultAsModule
... def foo():
...     return cube(2).up(10)
... 
>>> skin([foo(), foo()], z=[0, 1], slices=2)
[...includes...]
module foo(){
	up(z = 10) {
		cube(size = 2);
	}
}

skin(profiles = [foo(), foo()], slices = 2, z = [0, 1]);
```

but unfortunately this does also not work as parameter. The generated openscad code throws these errors:

```
WARNING: Ignoring unknown function 'foo' in file expsolid_out.scad, line 39
WARNING: Ignoring unknown function 'foo' in file expsolid_out.scad, line 39
ERROR: Assertion '(len(bad) == 0)' failed: "Profiles [0, 1] are not a paths or have length less than 3" in file ../../../.cache/pypoetry/virtualenvs/solidpython2-6LfNIStG-py3.10/lib/python3.10/site-packages/solid2/libs/BOSL2/skin.scad, line 409
```
So you need to somehow create a function that returns "something". If you figure out how to do it in OpenSCAD I can think about how it could be integrated into SolidPython.

ipsod, 2023-05-04:
Thanks Jeff.  That looks troublesome.  I'll look into it and let you know if I figure anything out.

`exportResultAsModule` is cool, by the way.

closed 2023-07-23 by jeff-dh