jeff-dh/SolidPython #32: Missing BOSL2 libraries.

j-d-salinger, 2023-06-02:


j-d-salinger, 2023-06-02:
Classic - it is in the github repo, but not the pypi release

closed 2023-06-02 by j-d-salinger

jeff-dh, 2023-06-02:
It's supposed to be there (in the pypi package)!

Which version are you using?

This is how it's supposed to be:

```python
>>> from solid2.extensions.bosl2.shapes3d import pie_slice
>>> pie_slice
<class 'solid2.extensions.bosl2.shapes3d.pie_slice'>
```

j-d-salinger, 2023-06-04:
Python 3.9.2, idk man

jeff-dh, 2023-06-04:
Which version of the SolidPython PyPi package (`pip show solidpython2`)

On Sun, 04 Jun 2023 07:45:57 -0700
j-d-salinger ***@***.***> wrote:

> Python 3.9.2, idk man
>

j-d-salinger, 2023-06-21:
It's gone... It can never come back.... Jk... I really don't want to break my system though by trying to re-install this.... If the drive crashes again and I have to reinstall all my dependencies, I'll come back here and update you