jeff-dh/SolidPython #34: Combine with nicescad

WolfgangFahl, 2023-07-29:
http://nicescad.bitplan.com/ allows to render openscad files and edit them in a browser (with pointers to many other similar solutions). I'd love to integrated SolidPython and nicescad to a convenient solution. I think things would not be to hard in nicescads local mode with full access to the local computer. For the server version i'd be a bit afraid to run python code which is supplied by user input.

jeff-dh, 2023-07-29:
Concerning sandboxing python code: are you aware of pyodide? Could it maybe solve the issue?

A very brief test with pyodide revealed that you can't import the current solid2 pypi package, because the platform is unknown to solid2, but when we disable the pickle cache and don't try to render the stl file with solid2, I assume it would work.

jeff-dh, 2023-07-29:
Hah, here is the current solidpython2 package from pypi running in pyodide (https://pyodide.org/en/latest/console.html) with a little hack ;)

```python
>>> import micropip
>>> await micropip.install("solidpython2")
>>> import platform
>>> platform.system = lambda: "Linux"
>>> from solid2 import *
>>> cube(2).left(10).up(5) - sphere(1).right(1).down(1)
difference() {
    translate(v = [0, 0, 5]) {
        translate(v = [-10, 0, 0]) {
            cube(size = 2);
        }
    }
    translate(v = [0, 0, -1]) {
        translate(v = [1, 0, 0]) {
            sphere(r = 1);
        }
    }
}
>>> 
```

jeff-dh, 2023-07-29:
I guess this is what you probably want:

https://gist.github.com/jeff-dh/c94b0727854432a014c1bdc24dad1c8a

WolfgangFahl, 2023-07-30:
@jeff-dh  - thank you for looking into this. Indeed i never tried https://pyodide.org/en/stable/ and trying out
```python
import micropip
await micropip.install("solidpython2")
import platform
platform.system = lambda: "Linux"
from solid2 import *
cube(2).left(10).up(5) - sphere(1).right(1).down(1)
```
in https://pyodide.org/en/latest/console.html immediately gave the output
```scad
difference() {
    translate(v = [0, 0, 5]) {
        translate(v = [-10, 0, 0]) {
            cube(size = 2);
        }
    }
    translate(v = [0, 0, -1]) {
        translate(v = [1, 0, 0]) {
            sphere(r = 1);
        }
    }
}
```
Installing nicescad or running openscad in the browser is obviously not possible. The issue of converting SolidPython code to openscad might be solveable with this approach if we e.g. make this a restful service in a docker instance.

WolfgangFahl, 2023-07-31:
The integration preparations are in the works. With the dockerized version at https://github.com/WolfgangFahl/nicescad/tree/main/docker you can now run a single docker compose and it will get you access to nicescad, openscad, blockscad, openjscad and a conversion page at the same time. How do we got from here? Currently most solidpython examples seem to mix the concerns of creation and rending. I'd rather have a clear definition of what  the design (mostly in variable d). The rendering of it e.g. in stl/png and it's openscad code is. I am currently thinking about providing the metadata for designg in some kind of json format. I think this is an important step toward providing proper "components" because i'd love to mix and match components from different sources especially openscad/blockscad/solidpython while openjscad seems not to be so integration friendly. Especially when looking at designs that are publicly available e.g. on thingiverse the component issue is important. I suggest we create a discussion/issue on the component issue somewhere. What would be your preffered place for this -here at nicescad or elsewhere and as issue or discussion / blog entry ...

jeff-dh, 2023-08-08:
I'll close this, because this seems to me like a nicescad issue and as such is a duplicate of WolfgangFahl/nicescad/issues/28

closed 2023-08-08 by jeff-dh