jeff-dh/SolidPython #37: Getting started guide

mraleson, 2023-08-01:
I think it would be nice if there was a simple getting started document.  When I found SolidPython took me a little effort work through the OpenSCAD, old SolidPython documentation, and new SolidPython2 code/examples.

Would there be any interest if I wrote a getting started tutorial of sort in markdown? If so, where is the best place to put it?  I could begin by creating a getting_started.md document?

jeff-dh, 2023-08-01:
Yeah, sure.....

I see / saw the README and the examples as getting started / tutorial, but I agree -- at least the README -- is a little messy and not well structured / organized and the examples might assume OpenSCAD knowledge.

I guess I'm fine with everything. Rewriting the README, adding and extra Getting Started document (and extracting the "getting started" part into it from the README),......
I guess it might make sense to have somehow two different "entry points" into solidpython. One for people already familiar with OpenSCAD and one for people who are not.

Intuitively I would suggest to split the README up into several documents. I would probably also extract the whole "Version 2.0" section.

mraleson, 2023-08-03:
I created a pull request #40

closed 2023-08-03 by mraleson