jeff-dh/SolidPython #39: BOSL2 extension side effects

mraleson, 2023-08-03:
Importing the BOSL2 extension has side effects, which might be a bad practice.  For example the functionality of `cylinder` unexpectedly changes after importing the BOSL2 extension.

Without extension `cylinder(2, 4)` create a cylinder and with the extension `cylinder(2, 4)` creates a cone.

jeff-dh, 2023-08-03:
Hmmmm this one is pretty bad....

To give you some insights: there are two sets of builtins, the regular openscad ones and the bosl2 ones. Importing bosl2 replaces the openscad builtins with the bosl2 ones.
There was a bug in the parameter list / order of the regular openscad cylinder builtin.

I fixed it and it now behaves according to the docs, but this causes some major backwards compatibility issues! This means the behaviour of the cylinder primitives changes for any call that uses positional parameters... I'm afraid this could break quite a few models out there.....

Probably I'll release a version called 2.1.0 soon which could contain this fix. I don't see any other solution to this yet.

I need to think about it for some time how to handle this.

jeff-dh, 2023-08-03:
Btw: thanks a lot for your great contributions! I really appreciate them, they push SolidPython a step further!

jeff-dh, 2023-08-03:
The alternative solution would be to adjust the BOSL2 cylinder primitive according to the "wrong" openscad builtin.

This would solve the issue, would hopefully not cause as many backwards compatibility issues, but would not be according to the [docs](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#cylinder)...... (and the wrong parameter order is actually more intuitive for me)

PR #42 (2023-08-08): Alternative fix cylinder parameters
fixes #39
closed 2023-08-08 by jeff-dh