jeff-dh/SolidPython #48: solid2/jupyterscad override render in the namespace

jreiberkyle, 2023-10-25:
The suggested way to use solid2 is `from solid2 import *`. One of the functions imported is `render`. This causes a conflict when using solid2 and jupyterscad together:

```python
from jupyterscad import render
from solid2 import *
```
Inspecting the `render` fcn that is in the namespace we see that the openscad render fcn is what is in the namespace:

```
Init signature: render(convexity: int = None) -> None
Source:        
class render(OpenSCADObject):
    """
    Always calculate the CSG model for this tree (even in OpenCSG preview
    mode).

    :param convexity: The convexity parameter specifies the maximum number of front sides (back sides) a ray intersecting the object might penetrate. This parameter is only needed for correctly displaying the object in OpenCSG preview mode and has no effect on the polyhedron rendering.
    :type convexity: int
    """

    def __init__(self, convexity: int = None) -> None:
        super().__init__('render', {'convexity': convexity})
File:           ~/.pyenv/versions/jupyterscad-dev/lib/python3.11/site-packages/solid2/core/builtin_primitives.py
Type:           type
Subclasses: 
```
But this function is useless outside of openscad GUI, since it just specifies the preview mechanics ([openscad user manual entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#render)).

What would you think of removing `render` from the solid2 namespace?

jeff-dh, 2023-10-25:
first of all some context:

#45
#44 
https://github.com/jeff-dh/SolidPython/wiki/Importing-SolidPython

I would say that using the openscad GUI is still the default workflow for solidpython2 and therefor it's useful in solidpython2.

in that case (which seems pretty unusual since we got the solidpython2-jupyterscad-integration) I would suggest to use this:

```python
from jupyterscad import render as jupyter_render
from solid2 import * # maybe also a different solid2 import style see https://github.com/jeff-dh/SolidPython/wiki/Importing-SolidPython

jupyter_render(render(cube()))
```

jeff-dh, 2023-10-25:
> The suggested way to use solid2 is `from solid2 import *`.

This change recently...... see the context links in the previous post.
The docs and the examples now use:

```python
from solid2 import cube, sphere, ...
```

And the wiki docs discuss three styles:

```python
from solid2 import *
from solid2 import cube, sphere
import solid2 as s
```

jreiberkyle, 2023-10-25:
oh good points. I'll continue forward with one of the last two suggestions above.

closed 2023-10-25 by jreiberkyle

jeff-dh, 2023-10-25:
> ```python
> from jupyterscad import render
> from solid2 import *
> ```

Btw: if you need this for some docs / example / show case for jupyterscad you could probably hack a solution by changing the order of the imports. I would assume a later import overwrites a previous one, so first importing solid2 and importing jupyterscad afterwards would result in exactly what you were asking for ;)