jeff-dh/SolidPython #51: How would one `include` external libraries?

dneise, 2023-11-02:
Hey there! 
I've not yet started with openScad, but am still looking around and found your awesome project.

I found that openScad supports external libraries, like e.g. these chamfered primitives https://github.com/SebiTimeWaster/Chamfers-for-OpenSCAD

In oder to use them in the openScad lang one needs to [`include <Chamfers-for-OpenSCAD/Chamfer.scad>;`](https://github.com/SebiTimeWaster/Chamfers-for-OpenSCAD/tree/master#usage). Now I wonder how and if this translates to Python?

For sure, we cannot simply do:
```python
from Chamfers-for-OpenSCAD import Chamfer

Chamfer.chamferCube(...)
```

Have a great day and thanks a lot!

jeff-dh, 2023-11-02:
https://github.com/jeff-dh/SolidPython/blob/master-2.0.0-beta-dev/solid2/examples/07-libs.x.py

Furthermore solidpython has a relative new feature which even supports `from chamfer_lib import Chamfer` checkout 
https://github.com/jeff-dh/SolidPython/blob/master-2.0.0-beta-dev/solid2/extensions/openscad_extension_generator.py

But I would suggest to take look at the built in bosl2 library first. It includes a whole lot of stuff (and is built in and the bosl2 support is pretty good I think).

jeff-dh, 2023-11-02:
[image]

dneise, 2023-11-06:
Thanks so much! I think we can close this

closed 2023-11-06 by dneise