jeff-dh/SolidPython #55: Openscad failed to include bosl2 files

elyoni, 2024-01-11:
Hi, 
I am working with:
- OpenSCAD 2024.01.03
- SolidPython 2.1.0
- Poetry 1.7.1

Yesterday I started working with Bosl2 from SolidPython2, and for some reason, OpenScad has issues with including the bosl2 files. The path is correct and, I can read the content of the files. But for some reason, I couldn't make it to work.

Some of the errors I see:
`WARNING: Can't open include file '/home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/version.scad'. in file ../../../../../.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/version.scad, line 1
WARNING: Can't open include file '/home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/constants.scad'. in file ../../../../../.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/constants.scad, line 2
WARNING: Can't open include file '/home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/transforms.scad'. in file ../../../../../.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/transforms.scad, line 3
WARNING: Can't open include file '/home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2
....
....
Compiling design (CSG Tree generation)...
WARNING: Ignoring unknown module 'cuboid' in file plate.scad, line 131
WARNING: Ignoring unknown module 'cuboid' in file plate.scad, line 136
WARNING: Ignoring unknown module 'cuboid' in file plate.scad, line 141
WARNING: Ignoring unknown module 'cuboid' in file plate.scad, line 146
`

* I didn't test it without poetry

Thanks For your help

jeff-dh, 2024-01-11:
Could you provide the (include section) of the generated scad file?

elyoni, 2024-01-11:
`include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/version.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/constants.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/transforms.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/distributors.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/mutators.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/color.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/attachments.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/shapes3d.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/shapes2d.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/drawing.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/masks3d.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/masks2d.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/math.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/paths.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/lists.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/comparisons.scad>;
include </home/yoni/.cache/pypoetry/virtualenvs/keyboardgenerator-OErBTrim-py3.10/lib/python3.10/site-packages/solid2/extensions/bosl2/BOSL2/linalg.scad>;
`

jeff-dh, 2024-01-11:
And those files really exist?

if that's the case I've no idea what the issue could be up until now....

elyoni, 2024-01-11:
Yes, they are. OK thanks

closed 2024-01-11 by elyoni

jeff-dh, 2024-01-11:
Did you solve it?

elyoni, 2024-01-11:
Yes. The issue was using Snap. Because it was running in the Snap environment, OpenScad couldn't reach for the scad files of Bosl2.
When I replaced the Installation using AppImage, it worded.
I apologize for your time, at first, I thought it's related to poetry and running in venv