jeff-dh/SolidPython #57: through no doc for import(center=...), here is an option exists.

FFengIll, 2024-02-03:
in openscad, import support a option named `center=True|False` in fact.

```py
class import_(OpenSCADObject):
   ...
```

jeff-dh, 2024-03-17:
Hmmmm since the docs do not mention it....... don't know....

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_Language#import

you could create an extension for yourself if you need that option

```python
from solid2.core.object_base import OpenSCADObject
from pathlib import Path

class my_import(OpenSCADObject):
    def __init__(self, file: PathStr, origin: P2 = (0, 0), convexity: int = None,
                        layer: int = None, center: bool = True) -> None:
        super().__init__('import',
                         {'file': Path(file).as_posix(), 'origin': origin,
                          'convexity': convexity, 'layer': layer, 'center': center})
```

jeff-dh, 2024-03-17:
PS: sorry for the late reply, did not spend any time on this project in the last couple of weeks.....

jeff-dh, 2024-03-17:
Or another solution:

```python
>>> from solid2 import import_
>>> i = import_("bla.stl")
>>> i._params["center"]=False
>>> i
import(center = false, file = "bla.stl", origin = [0, 0]);
```

FFengIll, 2024-03-18:
Thanks @jeff-dh

closed 2024-03-18 by FFengIll