jeff-dh/SolidPython #59: Help with features

bkpgh, 2024-05-24:
SolidPython2 seems great, but the lack of documentation is a real hindrance to using it. As an example, how do I use "extrude_along_path" ? "import solid2" doesn't seem to get it and there doesn't seem to be a solid2.utils. How do I use features like that that are available in solidpython?

Is solidpython2 under active development?

Thanks.

jeff-dh, 2024-05-24:
I'll reply in detail tomorrow or so.

To get you "started" with path extrude in solidpython2, take a look at this:

https://github.com/jeff-dh/SolidPython/blob/ce36f0651ecfddb04c34c36ebbd0727cd93541df/solid2/examples/07-libs-bosl2.py#L11

jeff-dh, 2024-05-25:
the documentation for solidpython2 is in the [wiki](https://github.com/jeff-dh/SolidPython/wiki) and the [examples](https://github.com/jeff-dh/SolidPython/tree/master-2.0.0-beta-dev/solid2/examples). You're right, the documentation is not great but I think it's reasonable. Everyone is welcome to contribute and improve the docs.....! ;)

take a look [here](https://github.com/jeff-dh/SolidPython/wiki/Version-2.x.x#backwards-compatibility) for documentation about backwards compatibility, `solid.utils` and `extrude_along_path`

I would say SolidPython2 is not under active development but it is maintained and in a solid state. From my point of view SolidPython2 seams to be in a pretty solid state and it seams to me as if it's actually in a "local minima".

If you post any problem concerning SolidPython2 I will read it and will probably reply soon and if there's a (fixable) bug, I will fix it, but there's no real development happening atm, if you are missing something we can discuss whether and / or how to "add" it.....

jeff-dh, 2024-06-03:
I'll close this for now.... reopen if you have any further issues / questions...

closed 2024-06-03 by jeff-dh