jeff-dh/SolidPython #61: How to create dimensioned drawings?

ToddG, 2024-07-14:
**WHAT**
I'd like to create dimensioned drawings for shop work.

**WHY**
For me, solidypython and openscad are tools for generating shop drawings, not for sending to a 3D printer.

**HOW**
I see that OpenSCAD has a library for creating dimensioned lines:

* https://cannymachines.com/entries/9/openscad_dimensioned_drawings
* https://github.com/sidorof/dimlines/tree/master

How can I use this openscad library in SolidPython?

For instance, here's one of the examples:

# cube.py
```
#! /usr/bin/env python

from solid2 import cube, sphere, difference, set_global_fn
from solid2 import import_scad
import_scad("dimlines.scad")

set_global_fn(72)

difference()(
    cube([100, 20, 30]),
    sphere(5)
).save_as_scad()
```

I have downloaded the `dimlines.scad` file to the same directory as the "cube.py" file above. How do I use the dimlines scad utilities to generate technical drawing lines etc?

jeff-dh, 2024-07-14:
- How to use external Openscad Libraries with SolidPython:
https://github.com/jeff-dh/SolidPython/blob/master-2.0.0-beta-dev/solid2/examples/07-libs.x.py

- How to use the dimlines library to create technical drawings:
https://cannymachines.com/entries/9/openscad_dimensioned_drawings

Something like this should be the solution (this is untested!):
```python
from solid2 impoprt scad_import

dimlines = scad_import("dimlines.scad")

dimlines.line(length=2,
                     width=dimlines.DIM_LINE_WIDTH,
                     height=dimlines.DIM_HEIGHT,
                     left_arrow=false, right_arrow=false).save_as_scad()
```

This could also work:

```python
from solid2 import use

use("dimlines.scad")

line(length=2, width=DIM_LINE_WIDTH, height=DIM_HEIGHT, left_arrow=false, right_arrow=false).save_as_scad()

```

jeff-dh, 2024-07-19:
I'll close this. if there are any further questions -> reopen or create a new ticket.

closed 2024-07-19 by jeff-dh