jeff-dh/SolidPython #64: Transfromer queing

mrVanDalo, 2024-08-02:
I love to use the operators with the assignment operator.

```python
fragment = cube([8, 50, thickness]).color("green")
fragment += cube([8, 13, thickness]).forward(50).color("blue")
fragment += cube([8 + 7, 10, thickness]).forward(50 + 13).left(7).color("orange")
part = fragment + fragment.mirror(1, 0, 0).right(16)
```

I would love to have something similar for transformations

```python
item = cube([width, height, thickness]).color("green")

cursor = right(width)
item += cursor(cube([thickness, height, depth])).color("blue")

cursor = cursor.up(depth)
item += cursor(cube([correction, height, thickness])).color("orange")

cursor = cursor.right(correction).back(correction)
item += cursor(part)
``` 

At the moment I have to fiddle around with lambdas and indices, which works, but is not so nice to read.

```python
item = cube([width, height, thickness]).color("green")

cursor1 = lambda x: x.right(width)
item += cursor1(cube([thickness, height, depth])).color("blue")

cursor2 = lambda x: cursor1(x).up(depth)
item += cursor2(cube([correction, height, thickness])).color("orange")

cursor3 = lambda x: cursor2(x).right(correction).back(correction)
item += cursor3(part)
```

jeff-dh, 2024-08-02:
What about this:

```python
from solid2 import *

def part():
    return cube(1).color("green") + subPart()

def subPart():
    return (cube(2).color("blue") + subsubPart()).right(1)

def subsubPart():
    return (cube(3).color("orange") + subsubsubPart()).up(2)

def subsubsubPart():
    return sphere(1).right(3).back(4)

print(part())
```

Isn't that basically what you want with another syntax?

jeff-dh, 2024-08-02:
Or probably even better like this:

```python
def part():
    return cube(1).color("green") + subPart().right(1)

def subPart():
    return cube(2).color("blue") + subsubPart().up(2)

def subsubPart():
    return cube(3).color("orange") + subsubsubPart().right(3).back(4)

def subsubsubPart():
    return sphere(1)
```

mrVanDalo, 2024-08-02:
Ich came up with this class for now.

```python
import functools
from solid2.core.object_base import AccessSyntaxMixin

class TransformationStack:
    """
    Gives the ability to stack transformations and act like a OpenSCAD Cursor

    Example

    ```python
    cursor = TransformationStack()
    item = cube([1,1,1])

    cursor.left(10)
    item += cursor.apply(cube([1,1,1]))

    cursor.up(10)
    item += cursor.apply(cube([1,1,1]))

    cursor.right(10)
    item += cursor.apply(cube([1,1,1]))
    ```
    """

    def __init__(self):
        self.stack = []

    def up(self, z):
        self.stack.append(lambda obj: obj.up(z))

    def down(self, z):
        self.stack.append(lambda obj: obj.down(z))

    def right(self, x):
        self.stack.append(lambda obj: obj.right(x))

    def left(self, x):
        self.stack.append(lambda obj: obj.left(x))

    def forward(self, y):
        self.stack.append(lambda obj: obj.forward(y))

    def back(self, y):
        self.stack.append(lambda obj: obj.back(y))

    def apply(self, obj: AccessSyntaxMixin):
        for operation in reversed(self.stack):
            obj = operation(obj)
        return obj
```

jeff-dh, 2024-08-02:
Well, but that's only a portion of it, right? What about rotation, scaling, hulls, colors,....? Not to mention the bosl2 transforms......

Your solution probably works, but I don't see any advantage over the function style solution I proposed above. Actually I think you don't even need the functions, you just need to compose (and think!) your object the other way around....

Btw: https://github.com/jeff-dh/SolidPython/blob/master-2.0.0-beta-dev/solid2/examples/06-functions.py

mrVanDalo, 2024-08-02:
Yeah, I haven't add all the functions yet, only the once I need so far.

The advantage is that it's easy to read from top to bottom. Which helps me a lot to model my objects.

create cube
move cursor to the right
create cube
move cursor up
create cube
move cursor to the right

closed 2024-11-19 by jeff-dh