jeff-dh/SolidPython #65: add support for manifold and openscad-nightly

bat52, 2024-09-10:
Manifold option allowing multi-threading, and resulting in significant performance improvements, has been integrated in openscad. 
https://github.com/openscad/openscad/issues/391#issuecomment-1718145488

It would be great if solidpython could make use of this option.
For instance:
`openscad --enable=manifold -o out.stl in.scad`

Additionally, while the feature is available since 2022, ubuntu official repos still deliver an openscad version that dates back to 2021.
Alternatively it is possible to install a nightly build snap
`snap install openscad-nightly`
but that requires running the command "openscad-nightly' instead of openscad.

jeff-dh, 2024-09-10:
please take a look at #53

bat52, 2024-09-11:
Here an example of how to add/modify class methods https://github.com/jeff-dh/SolidPython/issues/53

bat52, 2024-09-17:
Well in the end I realized I do not even know this, because I have a wrapper class, so I can export .scad file and covert to .stl with a custom command that I can define.
https://github.com/bat52/pylele/blob/main/api/sp2_api.py#L78

For the time being I am detecting the openscad version from shell call from which I can know if the manifold option is available.
I guess I could create an alias to redirect openscad calls to openscad-nightly, but I have not tried this yet.

Is there any performance loss in first exporting .scad, then exporting .stl , versus exporting directly in .stl?

jeff-dh, 2024-12-02:
5f2c987 should fix this issue 

```python
>>> from solid2.config import config
>>> config.openscad_stl_command = "openscad-nightly --enable=manifold -o {stlfile} {scadfile}"
>>> 
>>> from solid2 import cube
>>> cube(1).save_as_stl()
Traceback (most recent call last):
[...]
FileNotFoundError: [Errno 2] No such file or directory: 'openscad-nightly'
>>> 
```

reopen if there are any further issues.

closed 2024-12-02 by jeff-dh