jeff-dh/SolidPython #66: Output parametric SCAD files

sparr, 2024-10-02:
It's nice to be able to use python for control structures and other language features. However, once the SCAD file has been generated, I'd like to let other people (including the Thingiverse Customizer platform) change parameters of the model.

I'd like for the generated SCAD file to use SCAD variables and operations wherever possible. I know this would be very difficult/complex to implement, so I'm not expecting it to get implemented any time soon, but am hopeful that this issue will prompt some discussion about whether it's worthwhile, how it might work, pointers toward implementation, etc.

jeff-dh, 2024-10-05:
It's already implemented:

https://github.com/jeff-dh/SolidPython/blob/master-2.0.0-beta-dev/solid2/examples/10-customizer.py

sparr, 2024-10-05:
Oh, thanks, that looks like it would do most of what I was expecting.

I'd still hope for being able to do at least a loop, possibly more pythonic control structures, without having to scad_inline the whole thing.

This code replacing the scad_inline'd loop gets a little closer, allowing for solidpython / python code inside the loo instead of just literal scad codep, but is unwieldy / dangerous and produces not ideal indenting in the scad code:

```python
i = ScadVariable("i", 1)
scene += scad_inline("for (i = [1:objects]){")
scene += cube(side).translateX(2*i*side)
scene += scad_inline("}")
```

PS: is `union()` from that example the convention/standard/appropriate way to create a blank scene?

jeff-dh, 2024-10-06:
> [Verfolgungsbild][https://github.com/notifications/beacon/AAINZEMJIPS7SOY2ROT3AI3ZZ7GENA5CNFSM6AAAAABPIL3OKOWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUOYE6X6.gif]

I use union as an empty container. Solidpython(1) used to have part, which was removed and union serves as replacement.

Concerning the loop issue: it's technically not possible. Think about the different execution times. Python gets executed and produces the scad file. The scad file gets customized in openscad afterwards! There's no way for solidpython to access the customized variables....

sparr, 2024-10-06:
The example in my code shows that it is possible to do a SCAD loop that contains python-generated code. My approach is unsafe and unwieldy, but it does work.

I'm imagining an approach that might look something like this:

```python
i = ScadVariable("i", 1)
scad_for(i, range(1, objects+1), lambda:
    scene += cube(side).translateX(2*i*side)
)
```

This would require ~two new features:

1. `scad_for` which would emit the loop header, then run the function in its third parameter, then emit the trailing `}` of the loop
2. SCAD serialization of `range` objects.

Does that seem viable?

jeff-dh, 2024-10-06:
objects is your customized parameter? How do you want to "translate"
the objects object to the coresponding scad variable?

On Sun, 06 Oct 2024 04:40:05 -0700
sparr ***@***.***> wrote:

> The example in my code shows that it is possible to do a SCAD loop
> that contains python-generated code. My approach is unsafe and
> unwieldy, but it does work.
>
> I'm imagining an approach that might [...]

jeff-dh, 2024-10-06:
if you have any proposals or solutions feel free to go for it.

Either write an extension (if it's a couple of functions that can be
imported) or extend solidpython an propose a PR.

I could support you, but I don't think I have the time to take a
look at it myself atm.

sparr, 2024-10-06:
Sorry, I didn't make it explicit. My examples above are in the context of the customizer example you linked earlier. Here's a version that more clearly breaks out the python vs scad functionality and variable naming:

python:
```python
from solid2 import cube, text, translate, union, scad_render_to_file, scad_inline, \
                   CustomizerDropdownVariable, CustomizerSliderVariable, ScadVariable

#register all the custom variables you want to use
objects_var = CustomizerDropdownVariable("objects", 4, [2, 4, 6])
side_var = CustomizerSliderVariable("side", 4)

scene = union()

i_var = ScadVariable("i", 1)
one_var = ScadVariable("one", 1)
scene += scad_inline("for (")
scene += i_var
scene += scad_inline("=[")
scene += one_var
scene += scad_inline(":")
scene += objects_var
scene += scad_inline("]){")
scene += cube(side_var).translateX(2*i_var*side_var)
scene += scad_inline("}")

scad_render_to_file(scene)
```

openscad:
```openscad
objects = 4; //[2, 4, 6]
side = 4; //[]
i = 1;
one = 1;

union() {
	for (	i	=[	one	:	objects	]){	translate(v = [((2 * i) * side), 0, 0]) {
		cube(size = side);
	}
	}}
```

A "dumb" first draft of `scad_for` would do something like that, just outputting the appropriate scad source content around the parameters. In this case, I hand crafted the range expression.

PS: We'd probably also want to be able to serialize literals. `scene += 1` is currently an error because the python numeric type doesn't have a `_render()` method. My temporary workaround was to just put one in a variable.

jeff-dh, 2024-12-02:
What about this:

```python
def scad_for_test():
    def scad_range(start, stop, step=1):
        return f"[{start}:{step}:{stop}]"
    def scad_for(scadrange, loop_lambda):
        body = loop_lambda(ScadValue("scad_for_index"))._render()
        return scad_inline(f"for(scad_for_index = {scadrange})\n{{\n{indent(body)}\n}}\n")

    return scad_for(scad_range(0, 7), lambda i : cube(1).up(i * 2))
```

it seems to work

jeff-dh, 2024-12-02:
```python
def scad_if_test():
    def scad_if(condition, if_body, else_body):
        c = f"if({condition})\n{{\n{if_body._render()}\n}}\nelse\n{{\n{else_body._render()}\n}}"
        return scad_inline(c)

    c = cube(2).left(5)
    return scad_if(get_animation_time() < 0.5, c, c.color("red"))
```

jeff-dh, 2024-12-04:
fixed by [8fa8f64](https://github.com/jeff-dh/SolidPython/commit/8fa8f64aac96ff20c7c8b12b72c4affd240418d5)

```python
from solid2 import CustomizerDropdownVariable, \
                   CustomizerSliderVariable, \
                   cube, scad_range, scad_for, scad_if

number_of_cubes = CustomizerSliderVariable("number_of_cubes", 7)
alt_color = CustomizerDropdownVariable("alt_color", "red", ["red", "blue", "green"])

def f_loop(i):
    c = scad_if(i % 2 == 0, cube(), cube().color(alt_color))
    return c.up(i*2)

cubes_stacked = scad_for(scad_range(0, number_of_cubes-1), f_loop)

cubes_stacked.save_as_scad()
```

closed 2024-12-04 by jeff-dh

sparr, 2024-12-04:
Thanks! I look forward to giving these a try.