jeff-dh/SolidPython #67: Need a way to name entities

shreekumar3d, 2024-10-19:
I am looking for a way to get SolidPython to output a scad "module". I need this so that the scad output clearly shows the logic used to create the scene.  My scene consists of a booleans that operate on extruded polygons with lots of vertices.  Right now, the polygon coordinates are replicated.

As an example,
```
need_a_name = polygon([[0,0],[0,1],[0,1]])
s1 = need_a_name.linear_extrude(10)
s2 = need_a_name.linear_extrude(5).right(10)
s3 = s1+s2
print(s3)
```
would print replicate the polygon coordinates twice.

I am looking to achieve the following output
```
module need_a_name() {
        polygon(points = [[0, 0], [0, 1], [0, 1]]);
}
union() {
	linear_extrude(height = 10) {
                need_a_name();
	}
	translate(v = [10, 0, 0]) {
                need_a_name();        
	}
}
``` 

Any way this can be done in SolidPython2 ?

jeff-dh, 2024-10-19:
https://github.com/SolidCode/SolidPython/issues/197#issuecomment-1142865675

shreekumar3d, 2024-10-21:
Thanks @jeff-dh ! I added [[a comment](https://github.com/SolidCode/SolidPython/issues/197#issuecomment-2424904674)] in reply to SolidCode#197 . That issue is closed, but my comment provides better context there I feel.

closed 2024-11-19 by jeff-dh