jeff-dh/SolidPython #75: object.save_as_stl() hangs

ionburger, 2025-06-22:
from solid2 import cube, cylinder, set_global_fn

set_global_fn(100)

c = cylinder(r=13, h=40).rotate([0,90,0])
k = cube(0.5,0.5,0.5).translate([0,12.75,-0.25])
for i in range(50):
    k += k.translate([1,0,0])
c += k  
c.save_as_stl('knurl.stl')

still struggling to figure out exactly why this code causes a hang

jeff-dh, 2025-06-23:
I don't think it hangs, I think it renders forever.

1. resolution to high, start low poly and increase
2. are you sure about your loop? seems strange to me, maybe something like:

```python
for i in range(10):
    result += something.left(i*3)
```

you could also use `bosl2.distribute` or similar

ionburger, 2025-06-23:
Does your code do something different? i was under the impression left/right etc were just abstractions of translate. i did discover that it is definitely a performance issue, i was kind of expecting solid to use a scad loop but it appears to just manually copy/paste every iteration into the finished file, hence my performance issues

jeff-dh, 2025-06-23:
```python
from solid2.extensions.bosl2.shapes3d import cube, zcyl

k = zcyl(r=13, h=40)
c = cube(0.5).arc_copies(50, r=12.75).color("red")

(k+c).save_as_scad()
```

[image]